gazebosim/gz-sim #2217: System plugins keep executing after their model is removed.

jrutgeer, 2023-10-24:
## Environment
* OS Version: Ubuntu 22.04
* Harmonic, source built

## Description
Many system plugins need to be attached to a model (e.g. to access that model's links, collisions, etc).
However, when that model is removed, the plugin keeps running.
Given typically a model plugin needs  access entities of the model (that no longer exist), this leads to crashes.

## Expected behavior
When a model is removed, all of its plugins are removed as well, or at the least their pre/update/post methods are no longer called.

## Steps to reproduce
- Create a trivial system plugin, e.g. 

```c++
void TrivialPlugin::PreUpdate(const gz::sim::UpdateInfo &_info,
                         gz::sim::EntityComponentManager &_ecm)
{
  if (_info.paused) return;

  static int counter = 0;
  counter++;
  if (1000 == counter)
  {
    counter = 0;
    std::cout << "TrivialPlugin::PreUpdate" << std::endl;
  }
}
```

- Add this plugin to a model.

- Start the simulation

- Select the model in the Entity Tree, right click + 'remove'

- The model is removed, but the `cout` message is still being output to the terminal.

jrutgeer, 2023-10-25:
I implemented a simple workaround like this:

```c++
void WasteBin::PostUpdate(const UpdateInfo & _info, const EntityComponentManager & _ecm)
{
  if (_info.paused) return;
  if (!this->dataPtr->configured) return;

  if (!this->dataPtr->model.Valid(_ecm))
  {
    gzwarn << "Model <" << this->dataPtr->modelName << "> no longer valid. "
           << "Disabling WasteBin plugin." << std::endl;
    this->dataPtr->configured = false;
    return;
  }

  [...]

}
```

While this works well, it has following drawbacks:
- It is yet some additional code to implement in each model or sensor plugin; prone to be forgotten etc.
- I assume the call to `model.Valid(_ecm)` has a non-negligable computational cost, which would be avoided if a proper solution was implemented.

So I still feel that model and sensor plugins should be disabled (or removed) automatically upon removal of the corresponding model or sensor.

jrutgeer, 2023-10-25:
Another example:
- Add a model with some joints having a `JointPositionController` plugin,
- Remove that model

Result:

Flood of error messages:

`[Err] [EntityComponentManager.cc:1083] Trying to create a component of type [8319580315957903596] attached to entity [47], but this entity does not exist. This create component request will be ignored.`

</br>

I 'fixed' this in my local tree similar to above, but _meh_...

```c++
void JointPositionController::PreUpdate(
    const UpdateInfo &_info,
    EntityComponentManager &_ecm)
{
  GZ_PROFILE("JointPositionController::PreUpdate");

  if (kNullEntity == this->dataPtr->model.Entity())
  {
    return;
  }

  if (!this->dataPtr->model.Valid(_ecm))
  {
    gzwarn << "JointPositionController model no longer valid. "
           << "Disabling plugin." << std::endl;
    this->dataPtr->model = Model(kNullEntity);
    return;
  }

  [...]

}
```

arjo129, 2023-10-31:
Yes currently the `SystemManager` does not support removal of plugins. This seems to be something that can be fixed, but will take some amount of effort.

frede791, 2023-12-07:
Latching on to this: I get an `entity does not exist` error after removing a model that was spawned using the resource spawner. I think this is probably related to this issue. (Using gazebo garden)

arjo129, 2023-12-11:
Yeah. That is expected I'm currently working on a fix for it here: #2232

arjo129, 2024-07-17:
Fixed by #2232

PR #2232 (2024-07-17): Remove systems if their parent entity is removed
# 🦟 Bug fix

Fixes #2217

## Summary
In particular if a user despawns an entity, the associated plugin gets removed. This should prevent issues like #2165. TBH I'm not sure if this is the right way forward as a system should technically be able to access any entity in a traditional ECS.

[image]

The PR has now been reworked and greatly simplified. All we do is stop all worker threads if an entity is removed and then recreaye remaining threads.

## Checklist
- [x] Signed all commits for DCO
- [x] Added tests
- [ ] Updated documentation (as neede [... cut]
closed 2024-07-17 by arjo129