gazebosim/gz-sim #2229: :farmer: Error C2065: 'M_PI': undeclared identifier in AdvancedLiftDrag.cc(618,64)

Crola1702, 2023-11-06:
## Environment
* OS Version: Windows CI
* Source gz-sim build

## Description
* Expected behavior: Build should pass
* Actual behavior: Build is failing because of an undeclared identifier

## Steps to reproduce
1. Run a job in gz-sim7 or 8
2. See build fail

## Output
Reference build: [gz-sim8-win](https://build.osrfoundation.org/job/ign_gazebo-gz-8-win/28/consoleFull#12751072418ff58640-3599-4406-a210-216932f1748c)

Log output:
```
  gz-sim8-acoustic-comms-system.vcxproj -> C:\J\workspace\ign_gazebo-gz-8-win\ws\build\gz-sim8\bin\Release\gz-sim8-acoustic-comms-system.dll

  Building Custom Rule C:/J/workspace/ign_gazebo-gz-8-win/ws/gz-sim/src/systems/advanced_lift_drag/CMakeLists.txt

  AdvancedLiftDrag.cc

C:\J\workspace\ign_gazebo-gz-8-win\ws\gz-sim\src\systems\advanced_lift_drag\AdvancedLiftDrag.cc(618,64): error C2065: 'M_PI': undeclared identifier [C:\J\workspace\ign_gazebo-gz-8-win\ws\build\gz-sim8\src\systems\advanced_lift_drag\gz-sim8-advanced-lift-drag-system.vcxproj]

C:\J\workspace\ign_gazebo-gz-8-win\ws\gz-sim\src\systems\advanced_lift_drag\AdvancedLiftDrag.cc(674,46): error C2065: 'M_PI': undeclared identifier [C:\J\workspace\ign_gazebo-gz-8-win\ws\build\gz-sim8\src\systems\advanced_lift_drag\gz-sim8-advanced-lift-drag-system.vcxproj]
```

`M_PI` mentions: https://github.com/gazebosim/gz-sim/blob/gz-sim8/src/systems/advanced_lift_drag/AdvancedLiftDrag.cc#L618 and https://github.com/gazebosim/gz-sim/blob/gz-sim8/src/systems/advanced_lift_drag/AdvancedLiftDrag.cc#L674.

This is only failing in Windows

scpeters, 2023-11-06:
it should be `GZ_PI`

scpeters, 2023-11-06:
> it should be `GZ_PI`

from https://github.com/gazebosim/gz-math/blob/6460f7c0a59833e2729157d246ee07806ff56126/include/gz/math/Helpers.hh#L43-L55

PR #2230 (2023-11-06): Use `GZ_PI` instead of `M_PI` to fix windows builds
# 🦟 Bug fix

Fixes #2229

## Summary

Some of the `M_PI` usage was introduced in #2185, but I also found another in the marker example.

## Checklist
- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://gazebosim.org/docs/all/contributing#te [... cut]
commit 82fbdbab (2023-11-06): Use `GZ_PI` instead of `M_PI` to fix windows builds (#2230)

Fixes #2229
closed 2023-11-06 by azeey