gazebosim/gz-sim #2238: MecanumDrive system does not have a test

scpeters, 2023-11-11:
## Desired behavior

The existing system should be tested.

## Alternatives considered

## Implementation suggestion

Use the example world with the suggested commands in the XML comments at the top of the file.

* https://github.com/gazebosim/gz-sim/blob/gz-sim7/examples/worlds/mecanum_drive.sdf

## Additional context

PerFrivik, 2023-11-29:
I also have this issue, that It just does nothing.

PerFrivik, 2023-11-29:
Nevermind, I realized that the simulation was just not running, just press the orange play button on the bottom left. Or if you want  the simulation to run as soon as you launch gazebo, you need to add the `-r` flag which stands for run. 

gz sim -r mecanum_drive.sdf