gazebosim/gz-sim #2249: Python tests fail in CI on macOS

scpeters, 2023-11-17:
## Environment
* OS Version: macOS
* Source or binary build? binary builds of gz-sim7, gz-sim8

## Description
* Expected behavior: python tests should pass
* Actual behavior: python tests fail

gz-sim7: from https://build.osrfoundation.org/job/gz_sim-ci-gz-sim7-homebrew-amd64/4/

~~~
test 301
        Start 301: testFixture_TEST

301: Test command: /usr/local/Frameworks/Python.framework/Versions/3.12/bin/python3.12 "/Users/jenkins/jenkins-agent/workspace/gz_sim-ci-gz-sim7-homebrew-amd64/gz-sim/python/test/testFixture_TEST.py"
301: Working Directory: /Users/jenkins/jenkins-agent/workspace/gz_sim-ci-gz-sim7-homebrew-amd64/build/python
301: Environment variables: 
301:  PYTHONPATH=/usr/local/Cellar/gz-sim7/HEAD/lib/python/
301:  LD_LIBRARY_PATH=/usr/local/Cellar/gz-sim7/HEAD/lib:
301: Test timeout computed to be: 1500
301: Traceback (most recent call last):
301:   File "/Users/jenkins/jenkins-agent/workspace/gz_sim-ci-gz-sim7-homebrew-amd64/gz-sim/python/test/testFixture_TEST.py", line 21, in <module>
301:     from gz.math7 import Vector3d
301: ModuleNotFoundError: No module named 'gz.math7'
301/301 Test #301: testFixture_TEST ........................................***Failed    1.07 sec
~~~

gz-sim8: from https://build.osrfoundation.org/job/gz_sim-ci-gz-sim8-homebrew-amd64/4/

~~~
...
        Start 320: world_TEST

320: Test command: /usr/local/Frameworks/Python.framework/Versions/3.12/bin/python3.12 "/Users/jenkins/jenkins-agent/workspace/gz_sim-ci-gz-sim8-homebrew-amd64/gz-sim/python/test/world_TEST.py"
320: Working Directory: /Users/jenkins/jenkins-agent/workspace/gz_sim-ci-gz-sim8-homebrew-amd64/build/python
320: Environment variables: 
320:  PYTHONPATH=/usr/local/Cellar/gz-sim8/HEAD/lib/python/
320:  LD_LIBRARY_PATH=/usr/local/Cellar/gz-sim8/HEAD/lib:
320: Test timeout computed to be: 1500
320: Traceback (most recent call last):
320:   File "/Users/jenkins/jenkins-agent/workspace/gz_sim-ci-gz-sim8-homebrew-amd64/gz-sim/python/test/world_TEST.py", line 21, in <module>
320:     from gz_test_deps.math import SphericalCoordinates, Vector3d
320:   File "/Users/jenkins/jenkins-agent/workspace/gz_sim-ci-gz-sim8-homebrew-amd64/gz-sim/python/test/gz_test_deps/math.py", line 1, in <module>
320:     from gz.math7 import *
320: ModuleNotFoundError: No module named 'gz.math7'
320/320 Test #320: world_TEST ..............................................***Failed    0.76 sec
...
The following tests FAILED:
	305 - INTEGRATION_python_system_loader (Failed)
	313 - actor_TEST (Failed)
	314 - joint_TEST (Failed)
	315 - light_TEST (Failed)
	316 - link_TEST (Failed)
	318 - sensor_TEST (Failed)
	319 - testFixture_TEST (Failed)
	320 - world_TEST (Failed)
~~~

## Steps to reproduce

1. Run a macOS CI build
2.
3.

## Output

scpeters, 2023-11-17:
This may be related to which version of python3 is found by cmake and used to run the tests. Currently the gz-math and sdformat python bindings are built and installed for `python@3.11`, but gz-sim appears to find `python@3.12`, perhaps because it is a dependency of protobuf?

scpeters, 2023-11-20:
Looking at the protobuf formula in homebrew-core, I see that it supports multiple python versions:

* https://github.com/Homebrew/homebrew-core/blob/master/Formula/p/protobuf.rb#L25-L27
* https://github.com/Homebrew/homebrew-core/blob/master/Formula/p/protobuf.rb#L33-L37
* https://github.com/Homebrew/homebrew-core/blob/master/Formula/p/protobuf.rb#L64-L75

Crola1702, 2023-12-07:
Do you think this is related to [PythonSystemLoaderTest.LoadMultipleSystems](https://build.osrfoundation.org/job/gz_sim-ci-gz-sim8-homebrew-amd64/7/testReport/(root)/PythonSystemLoaderTest/LoadMultipleSystems/) test failure in [homebrew gz-sim8](https://build.osrfoundation.org/job/gz_sim-ci-gz-sim8-homebrew-amd64/7)?

azeey, 2023-12-13:
@scpeters, if protobuf depends on multiple python versions, is there a way to install pin a specific version of python for all our formulas?

scpeters, 2024-01-19:
> @scpeters, if protobuf depends on multiple python versions, is there a way to install pin a specific version of python for all our formulas?

Currently `python@3.11` is the most official version of homebrew python since it symlinks `/usr/local/bin/python3` to `python3.11`, so most of our homebrew formulae explicitly depend on `python@3.11` and install bindings to `lib/python3.11`, though the `gz-sim*` formulae do not.

Changing the `gz-sim*` formulae to use `python@3.11` explicitly may not be enough though since `python@3.12` is a direct dependency of `gz-sim*` via `gz-common*` and `gdal`, and I have observed that for systems that have both `python@3.11` and `python@3.12` installed, `cmake` appears to always find version 3.12. I haven't figured out why that is the case.

As I noted in https://github.com/gazebosim/gz-sim/issues/2249#issuecomment-1819690958, protobuf installs bindings for multiple versions of python, so that could be possible option for us, but it may be extra work.

scpeters, 2024-01-19:
> Changing the `gz-sim*` formulae to use `python@3.11` explicitly may not be enough though

I just tried updating `gz-sim8` to use `python@3.11` explicitly, but it still creates bindings for 3.12. I'll need to look into how it finds a particular version of python.

We could also just move everything to 3.12 and declare that anyone who wants to use python bindings needs to use 3.12

scpeters, 2024-01-23:
On a suggestion from @traversaro, in https://github.com/osrf/homebrew-simulation/pull/2540 I'm setting `Python3_EXECUTABLE` explicitly in cmake in order to use the default python version.

Crola1702, 2024-01-30:
Hey! I noticed the following log output in [gz-sim-main](https://build.osrfoundation.org/job/gz_sim-ci-main-homebrew-amd64/21/consoleFull#-394183657aa60f765-a60a-427d-900c-dc128ab22287) and [gz-launch-main](https://build.osrfoundation.org/job/gz_launch-ci-main-homebrew-amd64/16/consoleFull#-394183657aa60f765-a60a-427d-900c-dc128ab22287):

```
CMake Error at /usr/local/Cellar/cmake/3.28.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python3 (missing: Development Development.Module
  Development.Embed) (found version "3.12")
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.28.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/Cellar/cmake/3.28.2/share/cmake/Modules/FindPython/Support.cmake:3867 (find_package_handle_standard_args)
  /usr/local/Cellar/cmake/3.28.2/share/cmake/Modules/FindPython3.cmake:545 (include)
  CMakeLists.txt:151 (find_package)
```

I think this is related to 3.12 -> 3.11 changes, right?

scpeters, 2024-01-30:
> Hey! I noticed the following log output in [gz-sim-main](https://build.osrfoundation.org/job/gz_sim-ci-main-homebrew-amd64/21/consoleFull#-394183657aa60f765-a60a-427d-900c-dc128ab22287) and [...]

I think it will be fixed by https://github.com/gazebo-tooling/release-tools/pull/1110, but I'm still waiting for an approval on that PR

PR #2298 (2024-01-31): Explicitly check SKIP_PYBIND11 for python bindings
# 🦟 Bug fix

Part of https://github.com/gazebosim/gz-sim/issues/2249, uses https://github.com/osrf/homebrew-simulation/pull/2543 via `ci_matching_branch/`

## Summary

This is a minor change to the cmake logic for enabling python bindings and is used as an opportunity to test whether the changes in https://github.com/osrf/homebrew-simulation/pull/2543 will help fix #2249.

## Checklist
- [X] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Pyt [... cut]
closed 2024-01-31 by scpeters