gazebosim/gz-sim #2251: Windows warnings about inconsistent dll linkage when building runGui.exe

scpeters, 2023-11-21:
## Environment
* OS Version: Jenkins windows CI machines
* Source or binary build? source build of gz-sim7

## Description
* Expected behavior: build should compile without warnings
* Actual behavior: some windows compiler warnings
    * [image] https://build.osrfoundation.org/view/ign-garden/job/gz_sim-7-win/3/msbuild/

## Steps to reproduce

1. Run https://build.osrfoundation.org/view/ign-garden/job/gz_sim-7-win
2.
3.

## Output

~~~
  gz.cc

C:\J\workspace\gz_sim-7-win\ws\gz-sim\src\gz.cc(46,1): warning C4273: 'gzSimVersion': inconsistent dll linkage [C:\J\workspace\gz_sim-7-win\ws\build\gz-sim7\src\runGui.vcxproj]

C:\J\workspace\gz_sim-7-win\ws\gz-sim\src\gz.hh(24,36): message : see previous definition of 'gzSimVersion' [C:\J\workspace\gz_sim-7-win\ws\build\gz-sim7\src\runGui.vcxproj]

C:\J\workspace\gz_sim-7-win\ws\gz-sim\src\gz.cc(52,1): warning C4273: 'simVersionHeader': inconsistent dll linkage [C:\J\workspace\gz_sim-7-win\ws\build\gz-sim7\src\runGui.vcxproj]

C:\J\workspace\gz_sim-7-win\ws\gz-sim\src\gz.hh(28,36): message : see previous definition of 'simVersionHeader' [C:\J\workspace\gz_sim-7-win\ws\build\gz-sim7\src\runGui.vcxproj]

C:\J\workspace\gz_sim-7-win\ws\gz-sim\src\gz.cc(59,1): warning C4273: 'cmdVerbosity': inconsistent dll linkage [C:\J\workspace\gz_sim-7-win\ws\build\gz-sim7\src\runGui.vcxproj]

C:\J\workspace\gz_sim-7-win\ws\gz-sim\src\gz.hh(32,35): message : see previous definition of 'cmdVerbosity' [C:\J\workspace\gz_sim-7-win\ws\build\gz-sim7\src\runGui.vcxproj]

C:\J\workspace\gz_sim-7-win\ws\gz-sim\src\gz.cc(73,1): warning C4273: 'worldInstallDir': inconsistent dll linkage [C:\J\workspace\gz_sim-7-win\ws\build\gz-sim7\src\runGui.vcxproj]

C:\J\workspace\gz_sim-7-win\ws\gz-sim\src\gz.hh(35,42): message : see previous definition of 'worldInstallDir' [C:\J\workspace\gz_sim-7-win\ws\build\gz-sim7\src\runGui.vcxproj]

C:\J\workspace\gz_sim-7-win\ws\gz-sim\src\gz.cc(81,1): warning C4273: 'findFuelResource': inconsistent dll linkage [C:\J\workspace\gz_sim-7-win\ws\build\gz-sim7\src\runGui.vcxproj]

C:\J\workspace\gz_sim-7-win\ws\gz-sim\src\gz.hh(87,42): message : see previous definition of 'findFuelResource' [C:\J\workspace\gz_sim-7-win\ws\build\gz-sim7\src\runGui.vcxproj]

C:\J\workspace\gz_sim-7-win\ws\gz-sim\src\gz.cc(143,1): warning C4273: 'runServer': inconsistent dll linkage [C:\J\workspace\gz_sim-7-win\ws\build\gz-sim7\src\runGui.vcxproj]

C:\J\workspace\gz_sim-7-win\ws\gz-sim\src\gz.hh(62,34): message : see previous definition of 'runServer' [C:\J\workspace\gz_sim-7-win\ws\build\gz-sim7\src\runGui.vcxproj]

C:\J\workspace\gz_sim-7-win\ws\gz-sim\src\gz.cc(424,1): warning C4273: 'runGui': inconsistent dll linkage [C:\J\workspace\gz_sim-7-win\ws\build\gz-sim7\src\runGui.vcxproj]

C:\J\workspace\gz_sim-7-win\ws\gz-sim\src\gz.hh(80,34): message : see previous definition of 'runGui' [C:\J\workspace\gz_sim-7-win\ws\build\gz-sim7\src\runGui.vcxproj]

  gz.obj : MSIL .netmodule or module compiled with /GL found; restarting link with /LTCG; add /LTCG to the link command line to improve linker performance

     Creating library C:/J/workspace/gz_sim-7-win/ws/build/gz-sim7/lib/Release/runGui.lib and object C:/J/workspace/gz_sim-7-win/ws/build/gz-sim7/lib/Release/runGui.exp

  Generating code

  Finished generating code

  runGui.vcxproj -> C:\J\workspace\gz_sim-7-win\ws\build\gz-sim7\bin\Release\runGui.exe
~~~

scpeters, 2023-11-21:
Looking in the code history, I see that `gz.hh` was modified in https://github.com/gazebosim/gz-sim/pull/1911 (March 3, 2023) to make Garden windows CI greener. Then the `runGui` executable was added to citadel in https://github.com/gazebosim/gz-sim/pull/1962 (May 2, 2023) and merged forward to Garden in https://github.com/gazebosim/gz-sim/pull/2098 (August 29, 2023), which I believe is when these warnings were introduced.

mjcarroll, 2023-11-21:
I have a fix for this (that I thought I already applied).

mjcarroll, 2023-11-21:
Close by #2254

PR #2254 (2023-11-22): Fix DLL linkage/visibility issues
This was caused by rebuilding the same file as a library and standalone executable.  This simply separates the two.

Closes #2251
closed 2023-11-22 by scpeters