gazebosim/gz-sim #2266: WorldAngularVelocityCmd and WorldLinearVelocityCmd are ignored by Physics system

scpeters, 2023-12-12:
## Desired behavior

Currently the `AngularVelocityCmd` and `LinearVelocityCmd` components can be added to a model or link to set body-fixed velocity components (see [Physics.cc:2406-2608](https://github.com/gazebosim/gz-sim/blob/a88340b0d535905765aeaa41aab7b4d177fa6522/src/systems/physics/Physics.cc#L2406-L2608). Components are also defined for [WorldAngularVelocityCmd](https://github.com/gazebosim/gz-sim/blob/gz-sim7/include/gz/sim/components/AngularVelocityCmd.hh#L42-L47) and [WorldLinearVelocityCmd](https://github.com/gazebosim/gz-sim/blob/gz-sim7/include/gz/sim/components/LinearVelocityCmd.hh#L43-L49), but the Physics system ignores these components.

The Physics system should use these components so that velocities can be set in the world frame as an alternative to a body-fixed frame.

## Alternatives considered

## Implementation suggestion

This does raise a question what to do if an entity has both a `*VelocityCmd` and a `World*VelocityCmd` set; which one should be honored? Or should neither be honored?

I would suggest defining an order of precedence, perhaps preferring the body-fixed components since they were the first to be supported by the Physics system.

## Additional context

arjo129, 2023-12-14:
I suspect we should warn users if two plugins are setting `VelocityCmd` and `World*VelocityCmd`. My guess is the end user would be confused by the behavior if we don't warn them. We should have a defined ordering but at the very least by warning users, we would be preventing some degree of confusion.

On a related note, what happens if we set LinearVelocity* then try to apply a force? Or for instance the user sets a linear velocity and then tries to apply a force (it may be nice to support his at some point as it could clean up some of our tests)?

azeey, 2024-01-08:
Let's discuss whether we should be encouraging the use of `VelocityCmd` or creating a new component specifically for setting initial velocities.

scpeters, 2024-01-19:
> Let's discuss whether we should be encouraging the use of `VelocityCmd` or creating a new component specifically for setting initial velocities.

We currently use [JointPositionReset](https://github.com/gazebosim/gz-sim/blob/gz-sim8/include/gz/sim/components/JointPositionReset.hh) and [JointVelocityReset](https://github.com/gazebosim/gz-sim/blob/gz-sim8/include/gz/sim/components/JointVelocityReset.hh) to change joint states for a single timestep; so I suppose we could do the same with `WorldLinearVelocityReset` and `WorldAngularVelocityReset`.

traversaro, 2024-01-19:
> We currently use [JointPositionReset](https://github.com/gazebosim/gz-sim/blob/gz-sim8/include/gz/sim/components/JointPositionReset.hh) and [...]

My two cents: I recently started contributing to a project that uses ros2_control + Gazebo/gz, and the use of control/cmd naming for operations that are actually fiddling with the physics engine in a way that has no direct relation in the real world has been indeed a major source of confusion (see also https://discord.com/channels/1077825543698927656/1080763334384697430/1196933980252029099). So I am super in favor of using `Reset` as opposed to `Cmd` in this context.

fyi @AleTarsi