gazebosim/gz-sim #2267: Flickering sky when running world from sdf file.

frede791, 2023-12-12:
## Environment
* OS Version: 22.04
* Binary gz-garden latest release
* 
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre2]. 
      - [ ] Sensor rendering error. 
      - [x] GUI rendering error. 
      
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [x] running on real hardware

    - Rendering system info:
      - On Linux, provide the outputs of the following commands:

          LANG=C lspci -nn | grep VGA  
\00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:a7a0] (rev 04)

          echo "$DISPLAY"
":0"

          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'
                 direct rendering: Yes
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL vendor string: Intel
OpenGL renderer string: Mesa Intel(R) Graphics (RPL-P)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 23.0.4-0ubuntu1~22.04.1
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6 (Compatibility Profile) Mesa 23.0.4-0ubuntu1~22.04.1
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 23.0.4-0ubuntu1~22.04.1
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

          ps aux | grep Xorg
fremark+  198327  0.0  0.0   9536  2516 pts/18   S+   10:32   0:00 grep --color=auto Xorg

          sudo env LANG=C X -version
[ogre2.log](https://github.com/gazebosim/gz-sim/files/13646343/ogre2.log)

X.Org X Server 1.21.1.4
X Protocol Version 11, Revision 0
Current Operating System: Linux fremarkus-ThinkPad-X1-Carbon-Gen-11 6.1.0-1027-oem #27-Ubuntu SMP PREEMPT_DYNAMIC Mon Nov 20 15:08:14 UTC 2023 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.1.0-1027-oem root=UUID=87416edc-a5f8-4f3a-9a10-55881d31e973 ro quiet splash vt.handoff=7
xorg-server 2:21.1.4-2ubuntu1.7~22.04.2 (For technical support please see http://www.ubuntu.com/support) 
Current version of pixman: 0.40.0

    - [x] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering` 

<details>

Log is attached

</details>

## Description
* Expected behavior: The sky has a flickering in it.
* Actual behavior: The sky should not be flickering. 

## Steps to reproduce

1. Download the null island world from: https://app.gazebosim.org/OpenRobotics/fuel/worlds/Null%20Island
2. Start with `gz sim -r  path/to/null/island.sdf`

## Output
[image]

picture of flickering...

azeey, 2023-12-12:
@iche033 is this due to the non-NVIDIA GPU?

iche033, 2023-12-12:
I do remember seeing flickering sky issue on user laptops before. It could be a gpu / driver issue.

frede791, 2023-12-13:
Ok thanks. I will see if this error happens when i run it using a GPU.

biagio-trimarchi, 2024-01-02:
Hi! I am coming here from a [post](https://community.gazebosim.org/t/weird-flickering-when-cast-shadow-is-set-to-true/2452) I made on Gazebo Community.

This looks like the same problem I am facing. I wrote an SDF world with a sky and a directional light, and when I toggle the shadows the sky start flickering.

Here some information about my machine, as in the first post in this issue:
```Bash
$ LANG=C lspci -nn | grep VGA
00:02.0 VGA compatible controller [0300]: Intel Corporation Alder Lake-P GT1 [UHD Graphics] [8086:46a3] (rev 0c)
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation Device [10de:28a0] (rev a1)
```

```Bash
$ echo "$DISPLAY"
:0
```

```Bash
$ LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'
direct rendering: Yes
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL vendor string: Intel
OpenGL renderer string: Mesa Intel(R) Graphics (ADL GT2)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 23.0.4-0ubuntu1~22.04.1
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6 (Compatibility Profile) Mesa 23.0.4-0ubuntu1~22.04.1
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 23.0.4-0ubuntu1~22.04.1
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```

```Bash
$ ps aux | grep Xorg
biagio     26878  0.0  0.0   9212  2560 pts/6    R+   20:09   0:00 grep --color=auto Xorg
```

```Bash
$ sudo env LANG=C X -version

X.Org X Server 1.21.1.4
X Protocol Version 11, Revision 0
Current Operating System: Linux athena 6.2.0-39-generic #40~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov 16 10:53:04 UTC 2 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.2.0-39-generic root=UUID=bb022743-038b-409c-b381-21e78caa4017 ro quiet splash vt.handoff=7
xorg-server 2:21.1.4-2ubuntu1.7~22.04.5 (For technical support please see http://www.ubuntu.com/support) 
Current version of pixman: 0.40.0
```

frede791, 2024-01-03:
Yes. So I was able to confirm that if I use a GPU then the flickering does not appear.

iche033, 2024-01-03:
> I wrote an SDF world with a sky and a directional light, and when I toggle the shadows the sky start flickering.

oh sounds like the shadows (or the light) is causing the sky to flicker? One workaround would be to [disable shadows](http://sdformat.org/spec?ver=1.11&elem=light#light_cast_shadows) for the directional light.

frede791, 2024-01-06:
Yes this worked! If I disable the shadows for my sun and enable them for the scene only, I can run the sky.

frede791, 2024-01-16:
@iche033 Is having both sets of shadows enabled still a bug?

iche033, 2024-01-16:
> Is having both sets of shadows enabled still a bug?

I haven't seen any movement on this so I assume it's still an issue.

Rick-v-E, 2024-01-19:
Hi all,

I am running into the same issue. 

With shadows:
[image]

Without shadows, it is much better. However, still some flickering occurs:
[image]

I am also running it on a laptop without GPU:

```
$ LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'
direct rendering: Yes
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL vendor string: Intel
OpenGL renderer string: Mesa Intel(R) Graphics (ADL GT2)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 23.0.4-0ubuntu1~22.04.1
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6 (Compatibility Profile) Mesa 23.0.4-0ubuntu1~22.04.1
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 23.0.4-0ubuntu1~22.04.1
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```

ghfue, 2024-07-02:
I see this issue running on multiple computers with an Nvidia GPU. Disabling cast shadows from the sun also fixes the issue for me.