gazebosim/gz-sim #2281: Unable to load example worlds from main menu.

arjo129, 2024-01-10:
## Environment
* OS Version: Ubuntu 22.04
* Source:
  

</details>

## Description

When trying to load example worlds like the warehouse world from the quickstart menu, I get the error:
```
[Err] [Server.cc:198] Error Code 1: Msg: Error finding file [/home/arjoc/.gz/fuel/fuel.ignitionrobotics.org/openrobotics/models/warehouse/tip].
[Err] [Server.cc:198] Error Code 1: [/sdf/world[@name="world_demo"]/include[0]/uri:/home/arjoc/.gz/fuel/fuel.gazebosim.org/openrobotics/worlds/tugbot in warehouse/2/tugbot_warehouse.sdf:L176]: Msg: Unable to read file: [/home/arjoc/.gz/fuel/fuel.ignitionrobotics.org/openrobotics/models/warehouse/tip]
[Err] [Server.cc:198] Error Code 9: [/sdf/world[@name="world_demo"]:/home/arjoc/.gz/fuel/fuel.gazebosim.org/openrobotics/worlds/tugbot in warehouse/2/tugbot_warehouse.sdf:L3]: Msg: Error reading element <world>
[Err] [Server.cc:198] Error Code 9: Msg: Error reading element <sdf>
[Err] [Server.cc:198] Error Code 1: Msg: Unable to read file: [/home/arjoc/.gz/fuel/fuel.gazebosim.org/openrobotics/worlds/tugbot in warehouse/2/tugbot_warehouse.sdf]
```

* Expected behavior: The example world is loaded
* Actual behavior: 

## Steps to reproduce

1. Run `gz sim` on a clean install
2. Open menu. Click on the warehouse world example.

caguero, 2024-01-22:
I was able to reproduce it on `Harmonic`.

azeey, 2024-01-22:
This seems to be caused by having an `ignitionrobotics.org` uri. It can be reproduced by `gz sim -v4 "https://fuel.gazebosim.org/1.0/openrobotics/worlds/tugbot in warehouse"` as well without the quickstart menu.

mabelzhang, 2024-01-23:
Confirmed. If I replace `ignitionrobotics` with `gazebosim` in these 3 files
```
~/.gz/fuel/fuel.gazebosim.org/openrobotics/worlds/tugbot\ in\ warehouse/2/tugbot_warehouse.sdf
~/.gz/fue//fuel.gazebosim.org/movai/models/pallet_box_mobile/3/model.sdf
~/.gz/fuel/fuel.ignitionrobotics.org/movai/models/pallet/2/model.sdf
```
and reload the world
```
gz sim -v4 "https://fuel.gazebosim.org/1.0/openrobotics/worlds/tugbot in warehouse"
```
Then the world shows up.

To fix this, we need to upload new versions of these 3 models on Fuel, right? The ones under movai org, can we actually update them, or do we need to re-upload the models? The one under openrobotics we can probably update (I don't have permission to upload to that account).

arjo129, 2024-01-23:
Part of me wonders if we should batch run `sed` on all of these models to remove gazebosim. Or do something in the fuel frontend to give a warning if urls in fuel point to servers outside of fuel.

On a side note: If we are displaying models in the splash screen shouldn't we maintain our own fork of them? We definitely should acknowledge that the models come from the community but given that we may make breaking changes between versions this feels like something that can really hurt UX.

mabelzhang, 2024-01-23:
I thought that's what the code block I linked to above does, it seems to only keep the gazebosim.org and replaces ignitionrobotics.org . I think that's supposed to eliminate the need to fix all the models on Fuel, which there probably isn't an easy way because there are hundreds and they're all owned by different accounts.

(We are having a similar issue in DAVE right now, and I'm between keeping GitHub copies of the models and reuploading. We uploaded to Fuel specifically to make it easier to migrate to new Gazebo, but the models are all owned by different people, and I'd have to download, fix, re-upload to my account, and change all the URIs in the code. Alternatively, if we just keep a local copy on GitHub and forget about Fuel, then I don't have to worry about any of the reuploading and changing URIs.)

azeey, 2024-01-23:
This was a breaking change we made in Harmonic (https://github.com/gazebosim/gz-harmonic/blob/main/highlights.md#breaking-changes). I thought we updated all the models on Fuel to use gazebosim, so this is surprising. Maybe we didn't handle worlds? cc @nkoenig

mabelzhang, 2024-01-23:
So, 2 of the models in the warehosue world are not owned by openrobotics, but by movai. Only openrobotics ones have been updated, right?

1 of them is openrobotics though, and I had to change it locally.

mabelzhang, 2024-02-08:
Should... we just fix the movai models and reupload them to openrobotics org? Similarly with other broken models in the other splash screen worlds?

azeey, 2024-02-12:
> On a side note: If we are displaying models in the splash screen shouldn't we maintain our own fork of them? 

> Should... we just fix the movai models and reupload them to openrobotics org? Similarly with other broken models in the other splash screen worlds?

For the short term, I think fixing them and uploading them to openrobotics org with proper credits makes sense. For the longer term, I think we revert parts of https://github.com/gazebosim/gz-fuel-tools/pull/364/ and allow models that include "ignition" to work with a deprecation message.

azeey, 2024-03-18:
This should be fixed by https://github.com/gazebosim/gz-fuel-tools/pull/406