gazebosim/gz-sim #2285: gz-harmonic gui app is stuck at "[Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources"

rreddy80, 2024-01-12:
## Environment
* OS Version: Mac M1 Ventura 13.3.1
* Source or binary build?
  harmonic [brew install gz-harmonic]
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre | ogre2]. 
      - [ ] Sensor rendering error: n/a
      - [ ] GUI rendering error. [GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
    - EGL headless mode:
      - [ ] Running in EGL headless mode : n/a
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [x] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          echo "$DISPLAY"
          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          ps aux | grep Xorg
          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```
      - On Windows, run `dxdiag` and report the GPU-related information.
      - On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. Copy the output here.
        Graphics/Displays:

      ```
           Apple M1:
                  Chipset Model: Apple M1
                  Type: GPU
                  Bus: Built-In
                  Total Number of Cores: 8
                  Vendor: Apple (0x106b)
                  Metal Support: Metal 3
                  Displays:
                    Color LCD:
                      Display Type: Built-In Retina LCD
                      Resolution: 2560 x 1600 Retina
                      Main Display: Yes
                      Mirror: Off
                      Online: Yes
                      Automatically Adjust Brightness: Yes
                      Connection Type: Internal
```
    - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering`: n/a

<details>

```
# paste log here
```

</details>

## Description
* Expected behavior: gazebo opens up the world to run some tests
* Actual behavior: GUI requesting list of world names. The server may be busy downloading resources. Please be patient. is shown
[... 71 lines cut ...]
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.

azeey, 2024-01-26:
With the server running (`gz sim -s shapes.sdf`), can you try running the following from the terminal you tried running the GUI?
```
gz service -s /gazebo/worlds --reqtype gz.msgs.Empty --reptype gz.msgs.StringMsg_V -r 'unused: true' --timeout 2000
```
You should see:
```
data: "shapes"
```

If not, there might be something wrong in your network setup.

matt-mueller-ptc, 2024-01-29:
Chiming in to say that I have the same exact error when trying to get started with Gazebo. Working on Apple M2 Pro running Ventura 13.6.1.

@azeey when I try your instructions above, I get `Service call timed out` in the second terminal. How can I debug the network setup or know what the requirements are?

azeey, 2024-01-29:
I don't have much experience with network configuration on macOS, but if you can do it safely, a quick thing to try would be to disable the firewall and see if the issue goes away. In general, UDP multicast has been the main issue of these types of problems, so I recommend following my suggestion [here](https://github.com/gazebosim/gz-sim/issues/1879#issuecomment-1532288098)

vladveterok, 2024-05-26:
> ```
> gz service -s /gazebo/worlds --reqtype gz.msgs.Empty --reptype gz.msgs.StringMsg_V -r 'unused: true' --timeout 2000
> ```

Hey, gentlemen, I'm running these commands and receiving "data: "shapes", as mentioned in a quoted comment.
And yet when I'm running  `gz sim -v4 -g shapes.sdf` or any other .sdf file, I'm having this stack of countless 
`[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.`.
I use MacBook Pro M2 with macOS Sonoma 14.4.1

I'm struggling to find anything while googling. @matt-mueller-ptc did you manage to solve the problem? I appreciate any help.

Here's the whole stack:
```
[Msg] Gazebo Sim GUI    v8.3.0
[Dbg] [Application.cc:96] Initializing application.
[Dbg] [Application.cc:128] Qt using Metal graphics interface
[GUI] [Dbg] [Application.cc:657] Create main window
[GUI] [Wrn] [Application.cc:908] [QT] Populating font family aliases took 151 ms. Replace uses of missing font family "Roboto" with one that exists to avoid this cost.
[GUI] [Dbg] [PathManager.cc:68] Requesting resource paths through [/gazebo/resource_paths/get]
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[... cut ...]
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
```

IsraaAbdelghany9, 2024-10-03:
I have the same message but I using Ubuntu 22.04 only Gazebo11 classic GUI is working neither ignition nor Hurmonic I have deleted everything related to the classic version to  see if it is the problem but nothing happend the GUI of the harmonic appear which you can choose whic sdf file you want and when I hit run it fails and print the GUI is not responding and if I run it from the terminal it stays a little bit printing that kind of message wait message then it prints also the GUI is not responding ....... I found someone talking about the firewall and locale setting but nothing , also adding that line to the .bashrc -> export GAZEBO_RESOURCE_PATH=/usr/share/gazebo-harmonic
but also nothing happeded do anyone have any idea what is the problem ... 

when I run the comand too:

gz service -s /gazebo/worlds --reqtype gz.msgs.Empty --reptype gz.msgs.StringMsg_V -r 'unused: true' --timeout 2000

it prints out service time out

Mechazo11, 2024-10-20:
Hi I am also seeing this same issue for Harmonic built from source.

Mechazo11, 2024-10-20:
Hi @rreddy80 I found the solution to this problem as discussed here https://gazebosim.org/docs/latest/troubleshooting/

The following commands should resolve this problem. I tested it in Gazebo Harmonic source built in Ubuntu 22.04

```bash
sudo ufw allow in proto udp to 224.0.0.0/4
sudo ufw allow in proto udp from 224.0.0.0/4
gz sim -v 4 empty.sdf
```

Da-Dylan-Ma, 2024-10-21:
Hey @Mechazo11 That works for me as well, thanks! (Gazebo Ionic on Ubuntu 24.04)

So to me it looks like a problem caused by the firewall configs on one's local host, rather than an incompatibility among the versions or the OS or drivers etc..

Recommend to try this if any others ran into the same problem.

azeey, 2024-11-04:
I'll go ahead and close this since this is most likely a network configurations issue which has now been documented in https://gazebosim.org/docs/latest/troubleshooting/#network-configuration-issue

closed 2024-11-04 by azeey

hfcaio, 2024-12-04:
Hi @IsraaAbdelghany9 did you found a solution ? I've tried @Mechazo11 solution but it did not work.
I'm running ros2 and gazebo in a docker container, btw.

Mechazo11, 2024-12-04:
Hi @hfcaio which Gazebo are you using and have you tried source building it?

hfcaio, 2024-12-04:
Hi @Mechazo11.
 
I am using gazebo harmonic, and I tried both binary and source building. 

Like a said before, I'm running ros in a docker container, the base image is from [ardupilot](https://github.com/ArduPilot/ardupilot_dev_docker/blob/master/docker/Dockerfile_dev-ros). 
When I run `gz sim` I get:

[image]
but when I tried to open any world with `gz sim -v 4 empty.sdf` the result is [GUI] [Dbg] [Gui.cc:355] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.

[image]

Mechazo11, 2024-12-04:
@hfcaio can you post what you get when you echo ```GZ_SIM_RESOURCE_PATH``` after launching a container? For a simulator I worked on, this path needs to point to the directory where your world files are stored. An example is shown below

```bash
export GZ_VERSION=harmonic 
export GZ_SIM_RESOURCE_PATH=$GZ_SIM_RESOURCE_PATH:~/dev_ws/install/<project>_gz/share/<project>_gz/worlds
```

hfcaio, 2024-12-13:
Hi @Mechazo11, sorry for the delay.

[image]

And I am still having the same problem.

Mechazo11, 2024-12-13:
Hi @hfcaio this is what I would do 
0. Open a container and perform the following
1. Check if ```/usr/share/gz/gz-sim8/worlds``` has any thing in it? 
2. If yes, delete all ```build```, ```install``` folder and rebuild the workspace. Make sure no other packages other than what was defined by [Gazebo harmonic](https://github.com/Mechazo11/gazebo_harmonic_ws/blob/main/collection-harmonic.yaml). You can follow my source build example here: https://github.com/Mechazo11/gazebo_harmonic_ws. In my experience the key thing was to ensure ```merge-install``` was invoked when using building the workspace and no packages other than what was defined in the collection.yaml was installed.

PerFrivik, 2025-02-27:
> Hi [@rreddy80](https://github.com/rreddy80) I found the solution to this problem as discussed here https://gazebosim.org/docs/latest/troubleshooting/
> 
> The following commands should resolve this problem. I tested it in Gazebo Harmonic source built in Ubuntu 22.04
> 
> sudo ufw allow in proto udp to 224.0.0.0/4
> sudo ufw allow in proto udp from 224.0.0.0/4
> gz sim -v 4 empty.sdf

This fixed it for me, thank you sooooo much :DD