gazebosim/gz-sim #2289: Thread safety in custom rednering sensor system (DopplerVelocityLogSystem.cc)

5p00kk, 2024-01-17:
## Environment
## Description
In DopplerVelocityLogSystem.cc there are time-related variables: nextUpdateTime, simTime. They appear to be used by both: the rendering thread (OnRedner, PostRender) and physics thread (DoPostUpdate). There seems to be no thread safety for these variables, compared to for instance pendingRequests which is atomic.

azeey, 2024-01-26:
Thanks for catching those. There does seem to be a race condition on those variables. Would you be able to submit a PR?

sauk2, 2024-03-25:
@azeey Could I work on this?

azeey, 2024-03-25:
Yes, please go ahead. I'll assign it to you.

sauk2, 2024-03-26:
> Yes, please go ahead. I'll assign it to you.

Would like to know your opinion on a possible implementation idea here:

- Have a single mutex to protect both `simTime` and `nextUpdateTime`.
- Lock the mutex at the start of the [`OnRender`](https://github.com/gazebosim/gz-sim/blob/5944188294511a60b829f88851edeaded0b72a84/src/systems/dvl/DopplerVelocityLogSystem.cc#L583-L632) and [`OnPostRender`](https://github.com/gazebosim/gz-sim/blob/5944188294511a60b829f88851edeaded0b72a84/src/systems/dvl/DopplerVelocityLogSystem.cc#L635-L646) functions.
- [This](https://github.com/gazebosim/gz-sim/blob/5944188294511a60b829f88851edeaded0b72a84/src/systems/dvl/DopplerVelocityLogSystem.cc#L354-L405) `if` block in `DoPostUpdate` uses both variables. Break it down into a nested `if` and lock the mutex inside it.

PR #2345 (2024-03-28): Added mutex to protect stored time variables
# 🦟 Bug fix

Fixes #2289 

## Summary
Added a mutex to protect variables `simTime` and `nextUpdateTime` as they are used by both render and post update threads.

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://gazebosim.org/docs/all/contributing#test-coverage))

**Note to maintainers**: Remember to use **Squash-Merge** and edit the commit message to match the pull request sum [... cut]
closed 2024-03-28 by mjcarroll