gazebosim/gz-sim #2290: Custom sensor system example ignoring sdf's update rate

5p00kk, 2024-01-17:
## Description
In the implementation of a system for the custom sensor, the sensor is updated calling custom sensor's Update() method: https://github.com/gazebosim/gz-sim/blob/c158287d6f8556112d86d6e4188c39784364bd40/examples/plugin/custom_sensor_system/OdometerSystem.cc#L103
This will result in sensor update for at each simulation step and the system will ignore the sdf's update_rate attribute.

To avoid it the parent (gz::sensors::Sensor) Update() function should be called that executes the update at given update_rate (if force flag is false). For instance:

`
sensor->gz::sensors::Sensor::Update(_info.simTime, false);
`

5p00kk, 2024-01-19:
Probably it's two approaches as following call includes the same check (if forcing=False, the sensor will update itself only if it's time), the sensor's implementation is very similar to the guard in the system:

https://github.com/gazebosim/gz-sim/blob/c158287d6f8556112d86d6e4188c39784364bd40/src/systems/imu/Imu.cc#L171

https://github.com/gazebosim/gz-sensors/blob/d522779693916a81aac6360b049286e7ebb6cdb3/src/Sensor.cc#L456C4-L461C1

Regardless, the way OdometerSystem is implemented right now is incorrect because there is neither System level check or call to Sensor's class Update function that would check if it's time to update. Right now the direct Update implementation is being called.

EmmanuelMess, 2025-02-27:
Is there a workaround for this issue?

iche033, 2025-02-28:
if you're referring to the issue found in the custom odometry example, the change mentioned in https://github.com/gazebosim/gz-sim/issues/2290#issuecomment-1900693519 should make the sensor respect the specified update rate.

iche033, 2025-03-07:
not sure why you need to access `this->dataPtr->`? I updated the `custom_sensor_system` example in #2817 to respect sdf's update rate.

closed 2025-03-08 by iche033