gazebosim/gz-sim #2291: INTEGRATION_thruster has undefined behavior potentially causing test flakiness

azeey, 2024-01-17:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build? gz-sim7, 926923f6644ce1d2ef68316bb20432c88cd3b75e

## Description
* Expected behavior: Code has no undefined behaviors
* Actual behavior:  `std::vector::back` is potentially called on an empty vector in multiple places. This is regarded as [undefined behavior](https://en.cppreference.com/w/cpp/container/vector/back). For example:
https://github.com/gazebosim/gz-sim/blob/926923f6644ce1d2ef68316bb20432c88cd3b75e/test/integration/thruster.cc#L335-L340

The vector `modelPoses` is cleared followed by `auto latest_pose = modelPoses.back();`

GauravKumar9920, 2024-02-14:
can I work on this issue ?

azeey, 2024-02-14:
Yes, please!

azeey, 2024-02-15:
Hi @GauravKumar9920, this issue is not about code style, but that accessing `std::vector::back` on an empty vector could cause a crash.

PR #2350 (2024-04-02): Fixed undefined behavior in thruster.cc

# 🦟 Bug fix

Fixes #2291
## Summary
closed 2024-04-02 by arjo129