gazebosim/gz-sim #2299: `Gazebo/White` and other Materials that have a `light` pass appear very different from Gazebo classic's rendering

azeey, 2024-01-26:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build? gz-sim8
 
## Description
* Expected behavior:  Materials, especially commonly used ones, such as `Gazebo/White` to look reasonably close to Gazebo classic's rendering
* Actual behavior: 

**Gazebo classic**
[image]

**gz-sim8**
[image]

## Steps to reproduce

1. Run the following world
<details><summary>Details</summary>

```xml
<?xml version="1.0" ?>
<sdf version="1.6">
  <world name="shapes">
    <scene>
      <ambient>0.2 0.2 0.2</ambient>
      <background>0.8 0.8 0.8</background>
    </scene>

    <light type="directional" name="sun">
      <cast_shadows>true</cast_shadows>
      <pose>0 0 10 0 0 0</pose>
      <diffuse>0.8 0.8 0.8 1</diffuse>
      <specular>0.2 0.2 0.2 1</specular>
      <attenuation>
        <range>1000</range>
        <constant>0.9</constant>
        <linear>0.01</linear>
        <quadratic>0.001</quadratic>
      </attenuation>
      <direction>-0.5 0.1 -0.9</direction>
    </light>

    <model name="ground_plane">
      <static>true</static>
      <link name="link">
        <collision name="collision">
          <geometry>
            <plane>
              <normal>0 0 1</normal>
              <size>100 100</size>
            </plane>
          </geometry>
        </collision>
        <visual name="visual">
          <geometry>
            <plane>
              <normal>0 0 1</normal>
              <size>100 100</size>
            </plane>
          </geometry>
          <material>
            <ambient>0.8 0.8 0.8 1</ambient>
            <diffuse>0.8 0.8 0.8 1</diffuse>
            <specular>0.8 0.8 0.8 1</specular>
          </material>
        </visual>
      </link>
    </model>

    <model name="box">
      <pose>0 0 0.5 0 0 0</pose>
      <link name="box_link">
        <inertial>
          <inertia>
            <ixx>0.16666</ixx>
            <ixy>0</ixy>
            <ixz>0</ixz>
            <iyy>0.16666</iyy>
            <iyz>0</iyz>
            <izz>0.16666</izz>
          </inertia>
          <mass>1.0</mass>
        </inertial>
        <collision name="box_collision">
          <geometry>
            <box>
              <size>1 1 1</size>
            </box>
          </geometry>
        </collision>

        <visual name="box_visual">
          <geometry>
            <box>
              <size>1 1 1</size>
            </box>
          </geometry>
          <material>
            <script>
              <uri>file://media/materials/scripts/gazebo.material</uri>
              <name>Gazebo/White</name>
            </script>
          </material>
        </visual>
      </link>
    </model>
  </world>
</sdf>
```

</details>

iche033, 2024-01-26:
I think the current material parser just reads value from the 1st pass only and ignores other passes. 

`Gazebo/White` has 2 passes:

```
material Gazebo/White
{
  technique
  {
    pass ambient
    {
      ambient 1 1 1 1
    }

    pass light
    {
      diffuse 1 1 1 1
      specular .1 .1 .1 128
    }
  }
}
```

I actually don't see a reason to use 2 passes for this material. We could just combine them into one:

```
material Gazebo/White
{
  technique
  {
    pass ambient
    {
      ambient 1 1 1 1
      diffuse 1 1 1 1
      specular .1 .1 .1 128
    }
  }
}
```

iche033, 2024-01-26:
I think most materials with multiple passes could also be merged into a single pass. There are others, e.g. [Gazebo/TurquoiseGlowOutline](https://github.com/gazebosim/gz-sim/blob/08b5cfee79f680ce012c703de7f7b0ce965e7ff9/src/rendering/MaterialParser/gazebo.material#L536] that do seem like multiple passes are intended and may be more difficult to port.

azeey, 2024-01-26:
Yes, I think combining them manually in the `gazebo.material` file is probably the best way to go.

quarkytale, 2024-01-29:
I had the same thought, and since we're shipping a new file we can modify it however it fits best I guess. I don't mind taking this up and editing `gazebo.material` manually.

quarkytale, 2024-02-01:
> material Gazebo/White
> {
> . . .
>       specular .1 .1 .1 128
>     }

This 128 or any value > 1 should be divided by 256 right?

azeey, 2024-02-01:
Yes, I believe so: https://github.com/gazebosim/gz-math/blob/ec655a9260c1b1c5d3765c160999fd6e4d625788/include/gz/math/Color.hh#L281-L284

PR #2302 (2024-02-02): Fix Gazebo/White and refactored MaterialParser
# 🦟 Bug fix

Fixes #2299

## Summary

Edited gazebo.material file to remove multiple `pass`es  in solid colors.
Refactored the parser based on Gazebo's style guide.

## Checklist
- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://gazebosim.org/ [... cut]
closed 2024-02-02 by azeey