gazebosim/gz-sim #2318: Proposal: system for setting initial position / velocity

scpeters, 2024-02-15:
## Desired behavior

I'd like to be able to specify initial model/link velocity and initial joint position/velocity. For joints, I believe this can be done by setting [JointPositionReset](https://github.com/gazebosim/gz-sim/blob/gz-sim8/include/gz/sim/components/JointPositionReset.hh) and [JointVelocityReset](https://github.com/gazebosim/gz-sim/blob/gz-sim8/include/gz/sim/components/JointVelocityReset.hh) components programmatically. It would be consistent with this pattern to add `LinearVelocityReset` / `AngularVelocityReset` and/or `WorldLinearVelocityReset` / `WorldAngularVelocityReset` components that could be used to set initial model / link velocities.

## Alternatives considered

I had started an alternative approach in https://github.com/gazebosim/gz-sim/pull/2265 based on using the existing `AngularVelocityCmd` and `LinearVelocityCmd` components with the `VelocityControl` system, but I think the `*Reset` components make more sense.

## Implementation suggestion

~~~
<model name="example_model">
  <link name="parent_link" />
  <link name="child_link" />
  <joint name="multiaxis_joint" type="universal">
    <axis>
      <xyz>0 0 1</xyz>
    </axis>
    <axis2>
      <xyz>1 0 0</xyz>
    </axis2>
  </joint>
  <plugin name="SetInitialComponents" ...>
    <!-- different syntax ideas to set initial velocity for this model -->
      <!-- specify //entity/@type="model" and //entity/@name="__model__" -->
      <entity type="model" name="__model__">
        <set_component_value type="WorldLinearVelocityReset">1.0 4.0 9.0</set_component_value>
      </entity>
      <!-- specify //model/@name="__model__" -->
      <model name="__model__">
        <set_component_value type="WorldLinearVelocityReset">1.0 4.0 9.0</set_component_value>
      </model>
      <!-- or just treat set_component_value tags at the root scope as referring to the model -->
      <set_component_value type="WorldLinearVelocityReset">1.0 4.0 9.0</set_component_value>
  
    <!-- different syntax ideas to set initial velocity for a link -->
      <!-- specify //entity/@type="link" and //entity/@name="parent_link" -->
      <entity type="link" name="parent_link">
        <set_component_value type="WorldLinearVelocityReset">1.0 4.0 9.0</set_component_value>
      </entity>
      <!-- specify //link/@name="parent_link" -->
      <link name="parent_link">
        <set_component_value type="WorldLinearVelocityReset">1.0 4.0 9.0</set_component_value>
      </link>

    <!-- different syntax ideas to set initial position / velocity for a joint -->
      <!-- specify //entity/@type="joint" and //entity/@name="multiaxis_joint" -->
      <entity type="joint" name="multiaxis_joint">
        <set_component_value type="JointPositionReset">0.1 0.4</set_component_value>
        <set_component_value type="JointVelocityReset">1.0 -0.5</set_component_value>
      </entity>
      <!-- specify //joint/@name="multiaxis_joint" -->
      <joint name="multiaxis_joint">
        <set_component_value type="JointPositionReset">0.1 0.4</set_component_value>
        <set_component_value type="JointVelocityReset">1.0 -0.5</set_component_value>
      </joint>
  
  </plugin>
~~~

Thoughts are welcome for the right syntax here. I think we would have to hard-code a relationship between the `//set_component_value/@type` string and a `gz::sim::components::` class since the component Factory doesn't allow constructing just from a `string`. It does have [namesById and runtimeNamesById](https://github.com/gazebosim/gz-sim/blob/gz-sim8/include/gz/sim/components/Factory.hh#L464-L470) data structures but not the inverse (`IdsByName`). Even if we did, I'm not sure how it could be implemented.

I suggested the system name of `SetInitialComponents` in this example to go along with the `<set_component_value>` elements, but we could also use names with more specific semantics (similar to the [InitialVelocityPlugin from gazebo-classic](https://github.com/gazebosim/gazebo-classic/blob/gazebo11/worlds/initial_velocity.world#L48-L51)).

## Additional context

scpeters, 2024-02-16:
consider expressing the model state using the syntax of the [model_state](http://sdformat.org/spec?ver=1.11&elem=state#state_model)

arjo129, 2024-02-17:
This would be really useful for reducing our test times. In particular we could use it to test plugins like the hydrodynamics/LiftDrag etc.