gazebosim/gz-sim #2329: [Bug] surface_vehicles.md workspace wget command not working

PerFrivik, 2024-03-06:
## Environment
* Ubuntu 22.04: 
* Source (gz-sim8, 2ca64d032) 
  
  

<details>

```
per@per ~ $ mkdir ~/gazebo_maritime_ws
wget https://raw.githubusercontent.com/gazebosim/gz-sim/gz-sim8/tutorials/files/surface_vehicles/gazebo_maritime_ws -o ~/gazebo_maritime_ws -r
wget: /home/per/gazebo_maritime_ws: Is a directory

```

</details>

## Description
* Expected behavior: The command from the surface_vehicles.md page to work. 
* Actual behavior: The error above, in the log section. 

## Steps to reproduce

Run this command: 

mkdir ~/gazebo_maritime_ws
wget https://raw.githubusercontent.com/gazebosim/gz-sim/gz-sim8/tutorials/files/surface_vehicles/gazebo_maritime_ws -o ~/gazebo_maritime_ws -r

## Output
Check the log section above. 

Thanks!

caguero, 2024-03-06:
Thanks for reporting it, see #2330 .

PerFrivik, 2024-03-07:
Nice, thanks for the fast response!

PerFrivik, 2024-03-07:
I got around the wget issues, but now when trying to run the sydney_regatta, I get the following error: 

[Err] [SystemLoader.cc:119] Failed to load system plugin: (Reason: Could not find shared library)
- Requested plugin name: [maritime::PublisherPlugin]
- Requested library name: [libPublisherPlugin.so]
- Library search paths:
  - /home/per/.gz/sim/plugins/
  - /usr/lib/x86_64-linux-gnu/gz-sim-7/plugins/
  - /home/per/.ignition/gazebo/plugins/
[Err] [SystemLoader.cc:119] Failed to load system plugin: (Reason: Could not find shared library)
- Requested plugin name: [maritime::WaveVisual]
- Requested library name: [libWaveVisual.so]
- Library search paths:
  - /home/per/.gz/sim/plugins/
  - /usr/lib/x86_64-linux-gnu/gz-sim-7/plugins/
  - /home/per/.ignition/gazebo/plugins/

I tried exporting the gazebo plugin paths to the binaries, but it still did not work. 

export GAZEBO_PLUGIN_PATH=/home/per/gz-sim/tutorials/files/surface_vehicles/gazebo_maritime_ws/install/lib:${GAZEBO_PLUGIN_PATH}