gazebosim/gz-sim #2338: Mismatch about default collision detector between SDFormat specification and Gazebo Sim documentaion

xela-95, 2024-03-19:
Hi,

I noted that the SDF documentation about the default collision detector is different from the Gazebo sim one:
- SDF says `fcl`: http://sdformat.org/spec?ver=1.11&elem=physics#dart_collision_detector
- Gazebo sim says `ode`: https://github.com/gazebosim/gz-sim/blob/6cfcd07c346a72546963ed56ade48daa828c2072/tutorials/physics.md?plain=1#L100 (also used in unit tests https://github.com/gazebosim/gz-physics/blob/7d30e47fb61fe4db9cba460717a3910df9deabee/dartsim/src/WorldFeatures_TEST.cc#L87)

traversaro, 2024-03-19:
My guess is that "fcl" is the Gazebo Classic default, while "ode" is the modern gz-sim default, but I did not double checked. 

xref to a kind of generic issue related to this: https://github.com/gazebosim/sdformat/issues/508

azeey, 2024-05-06:
From Ian: Maybe we should at least print a warning that the default is being overridden in gz-sim

azeey, 2024-05-06:
Another thought is that we have a dedicated page in the gz-sim documentation that indicates which SDF tags gz-sim actually supports.