gazebosim/gz-sim #2353: Boundingbox on Link API

jfelixre, 2024-04-02:
There is an alternative of Gazebo-classic's Link API Boundingbox function for ignition?

API reference page still shows a "TODO"

thanks!

azeey, 2024-05-06:
We already have the ability to get the bounding box, but it's through a component. Documenting this would be a good step as well.

gabrielfpacheco, 2025-02-19:
> We already have the ability to get the bounding box, but it's through a component. Documenting this would be a good step as well.

@azeey, isn't that only available to get the model bounding box, since the Physics plugin set it [here](https://github.com/gazebosim/gz-sim/blob/gz-sim9/src/systems/physics/Physics.cc#L2887-L2892)?

I couldn’t find any source that actually sets a `gz::components::AxisAligned` for a link entity. Therefore, as I had to get the bounding box of the link for a system plugin I developed, I created #2778 and https://github.com/gazebosim/sdformat/pull/1540 to provide such an API.

Please let me know if I'm missing something, I hope this could be helpful to others.

iche033, 2025-02-19:
I think the alternative approach would be to update the the physics system populate the bounding box component for links as well? I see the API available in gz-physics, [GetLinkBoundingBox](https://github.com/gazebosim/gz-physics/blob/gz-physics8/include/gz/physics/GetBoundingBox.hh#L40)

gabrielfpacheco, 2025-02-26:
> I think the alternative approach would be to update the the physics system populate the bounding box component for links as well

Yes, I think this would probably be more straightforward to implement. If I understood it correctly, this feature should be available for all currently supported physics engines, right?

I initially thought not to update the physics plugin because I did not want to add extra load to a critical plugin if this was not an actual requirement. However, I believe we could the same as for the model and only update the link entities containing the corresponding components.

Maybe I could update https://github.com/gazebosim/gz-sim/pull/2787 to add this feature to the physics system and update the link API so that it only computes from the SDF if an `AxisAlignedBox` component does not exist for the link entity. Or I could completely remove the calculation via SDF if you think it is more adequate.

Please, let me know what would be the best path forward.

iche033, 2025-02-28:
I think getting the bounding box info from the physics system is preferred because it has the information already, and there is no need for a custom mesh bounding box calculator. In the rare case the the users does not load a physics system in Gazebo (or a particular physics engine does not support the bounding box feature), we could then fall back to using the SDF API.

I think the first step is to implement a [similar logic for populating a model AxisAlignedBox](https://github.com/gazebosim/gz-sim/blob/7eef2303c6096bb8b547439239e81d5c8571234e/src/systems/physics/Physics.cc#L2855-L2895) for links in the physics system. The expected usage here is that plugin writers would need to manually create the `AxisAlignedBox` component for a Link, the physics system then populates it.

We can then update the Link class to add the `AxisAlignedBox` API like you did in https://github.com/gazebosim/gz-sim/pull/2787, except we would be the one creating the `AxisAlignedBox` component for the link in that class. I think this could be done through an `EnableBoundingBoxChecks` call (similar to `EnableVelocityChecks`). 

If you're ok with this approach, could you do the changes in the physics system in a separate PR? Mainly just to make the review process easier. Thanks for the contributions so far!

gabrielfpacheco, 2025-03-06:
> If you're ok with this approach, could you do the changes in the physics system in a separate PR? Mainly just to make the review process easier

Yeah, I like the suggested approach. As soon as I have some free cycles, I will make the changes to the open PR and open a new one for the physics.

> Thanks for the contributions so far!

My pleasure :)

PR #2787 (2025-06-18): Add support to get link AABB from its collisions
# 🎉 New feature

- Closes #2353
- Requires
  - https://github.com/gazebosim/sdformat/pull/1547

## Summary

This PR adds AABB support to the Link API. It computes the overall link aligned bounding box (in local and world coordinates) by merging all collision bounding boxes obtained from their geometry shapes. Plus, it adds utility functions to transform AABB coordinates and calculate mesh AABB using utilities from `gz::common::Mesh`.

## Test it
- Checkout the appropriate `gz-sim` and `sdformat` branches
- Build these packages
```console
$ colco [... cut]
PR #2963 (2025-07-07): Add support to get link AABB from its collisions (backport #2787)
# 🎉 New feature

- Closes #2353
- Requires
  - https://github.com/gazebosim/sdformat/pull/1547

## Summary

This PR adds AABB support to the Link API. It computes the overall link aligned bounding box (in local and world coordinates) by merging all collision bounding boxes obtained from their geometry shapes. Plus, it adds utility functions to transform AABB coordinates and calculate mesh AABB using utilities from `gz::common::Mesh`.

## Test it
- Checkout the appropriate `gz-sim` and `sdformat` branches
- Build these packages
` [... cut]