gazebosim/gz-sim #2356: Crash on opening world with Plasma Wayland session

seasonedfish, 2024-04-04:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build? Binary, 6.16.0

<details>

```
QSocketNotifier: Can only be used with threads started with QThread
[GUI] [Err] [Ogre2RenderEngine.cc:989]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /build/ogre-next-UFfg83/ogre-next-2.2.5+dfsg3/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 163)
[GUI] [Err] [Ogre2RenderEngine.cc:989]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /build/ogre-next-UFfg83/ogre-next-2.2.5+dfsg3/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 163)
[GUI] [Err] [Ogre2RenderEngine.cc:989]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /build/ogre-next-UFfg83/ogre-next-2.2.5+dfsg3/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 163)
[GUI] [Err] [Ogre2RenderEngine.cc:989]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /build/ogre-next-UFfg83/ogre-next-2.2.5+dfsg3/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 163)
[GUI] [Err] [Ogre2RenderEngine.cc:989]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /build/ogre-next-UFfg83/ogre-next-2.2.5+dfsg3/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 163)
[GUI] [Err] [Ogre2RenderEngine.cc:989]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /build/ogre-next-UFfg83/ogre-next-2.2.5+dfsg3/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 163)
[GUI] [Err] [Ogre2RenderEngine.cc:989]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /build/ogre-next-UFfg83/ogre-next-2.2.5+dfsg3/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 163)
[GUI] [Err] [Ogre2RenderEngine.cc:989]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /build/ogre-next-UFfg83/ogre-next-2.2.5+dfsg3/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 163)
[GUI] [Err] [Ogre2RenderEngine.cc:989]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /build/ogre-next-UFfg83/ogre-next-2.2.5+dfsg3/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 163)
[GUI] [Err] [Ogre2RenderEngine.cc:989]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /build/ogre-next-UFfg83/ogre-next-2.2.5+dfsg3/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 163)
[GUI] [Err] [Ogre2RenderEngine.cc:997] Unable to create the rendering window after [11] attempts.
[GUI] [Err] [Ogre2RenderEngine.cc:915] Failed to create dummy render window.
[GUI] [Err] [Ogre2RenderEngine.cc:916] Please see the troubleshooting page for possible fixes: https://gazebosim.org/docs/fortress/troubleshooting
Stack trace (most recent call last) in thread 286454:
#31   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x78c50a294ac2, in 
#30   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x78c5038ccca0, in 
#29   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x78c5038cbaf1, in QThread::exec()
#28   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x78c503ab875a, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#27   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x78c503b130b7, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#26   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x78c5037193e2, in g_main_context_iteration
#25   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x78c503771257, in 
#24   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x78c50371bd3a, in g_main_context_dispatch
#23   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x78c503b13a66, in 
#22   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x78c503abcf26, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
#21   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x78c503ab9e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#20   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x78c50316c712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#19   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x78c503ae741d, in QObject::event(QEvent*)
#18   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/gui/libGzScene3D.so", at 0x78c4ee27eddf, in ignition::gazebo::v6::RenderThread::RenderNext(ignition::gazebo::v6::RenderSync*)
#17   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/gui/libGzScene3D.so", at 0x78c4ee279ad8, in ignition::gazebo::v6::IgnRenderer::Initialize[abi:cxx11]()
#16   Object "/lib/x86_64-linux-gnu/libignition-gazebo6-rendering.so.6", at 0x78c4ee15cc6c, in ignition::gazebo::v6::RenderUtil::Init()
#15   Object "/lib/x86_64-linux-gnu/libignition-rendering6.so.6", at 0x78c4ee06080b, in ignition::rendering::v6::RenderEngineManager::Engine(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#14   Object "/lib/x86_64-linux-gnu/libignition-rendering6.so.6", at 0x78c4ee0605ee, in ignition::rendering::v6::RenderEngineManagerPrivate::Engine(EngineInfo, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#13   Object "/lib/x86_64-linux-gnu/libignition-rendering6.so.6", at 0x78c4ee069381, in ignition::rendering::v6::BaseRenderEngine::Init()
#12   Object "/usr/lib/x86_64-linux-gnu/ign-rendering-6/engine-plugins/libignition-rendering-ogre2.so", at 0x78c4899ccb52, in ignition::rendering::v6::Ogre2RenderEngine::InitImpl()
#11   Object "/usr/lib/x86_64-linux-gnu/ign-rendering-6/engine-plugins/libignition-rendering-ogre2.so", at 0x78c4899cbdb7, in ignition::rendering::v6::Ogre2RenderEngine::InitAttempt()
#10   Object "/lib/x86_64-linux-gnu/libOgreNextMain.so.2.2.5", at 0x78c4894d6738, in Ogre::ResourceGroupManager::initialiseAllResourceGroups(bool)
#9    Object "/lib/x86_64-linux-gnu/libOgreNextMain.so.2.2.5", at 0x78c4894e2f2e, in Ogre::ResourceGroupManager::parseResourceGroupScripts(Ogre::ResourceGroupManager::ResourceGroup*)
#8    Object "/lib/x86_64-linux-gnu/libOgreNextMain.so.2.2.5", at 0x78c48951c31c, in Ogre::ScriptCompilerManager::parseScript(Ogre::SharedPtr<Ogre::DataStream>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#7    Object "/lib/x86_64-linux-gnu/libOgreNextMain.so.2.2.5", at 0x78c489514112, in Ogre::ScriptCompiler::compile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char [... cut ...]
* Expected behavior: Gazebo doesn't crash
* Actual behavior: Gazebo crashes

## Steps to reproduce

1. Run gazebo with `ign gazebo`
2. Open "Prius On Sonoma Raceway"

## Output
N/A

## Other info
This only happens in a Plasma Wayland session. GNOME Wayland works, and Plasma X11 works as well.

I found https://github.com/gazebosim/gz-sim/issues/1934 and tried `env -u WAYLAND_DISPLAY ign gazebo`, but I get the same error.

- plasma-desktop version `4:5.24.7-0ubuntu0.1`
- plasma-workspace-wayland version `4:5.24.7-0ubuntu0.1`

seasonedfish, 2024-04-06:
Just tried the workarounds suggested in https://github.com/gazebosim/gz-sim/issues/1934#issuecomment-1931215527 and they worked for me, although ideally Gazebo should work without the user having to apply these.

gmmyung, 2024-11-06:
It seems like it still isn't fixed. I am experiencing the same issue in Ubuntu 24.04/ROS2 Jazzy.

```
[GUI] [Err] [Ogre2RenderEngine.cc:1301]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at ./.obj-x86_64-linux-gnu/gz_ogre_next_vendor-prefix/src/gz_ogre_next_vendor/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 165)
[GUI] [Err] [Ogre2RenderEngine.cc:1301]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at ./.obj-x86_64-linux-gnu/gz_ogre_next_vendor-prefix/src/gz_ogre_next_vendor/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 165)
[GUI] [Err] [Ogre2RenderEngine.cc:1301]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at ./.obj-x86_64-linux-gnu/gz_ogre_next_vendor-prefix/src/gz_ogre_next_vendor/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 165)
[GUI] [Err] [Ogre2RenderEngine.cc:1301]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at ./.obj-x86_64-linux-gnu/gz_ogre_next_vendor-prefix/src/gz_ogre_next_vendor/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 165)
[GUI] [Err] [Ogre2RenderEngine.cc:1301]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at ./.obj-x86_64-linux-gnu/gz_ogre_next_vendor-prefix/src/gz_ogre_next_vendor/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 165)
[GUI] [Err] [Ogre2RenderEngine.cc:1301]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at ./.obj-x86_64-linux-gnu/gz_ogre_next_vendor-prefix/src/gz_ogre_next_vendor/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 165)
[GUI] [Err] [Ogre2RenderEngine.cc:1301]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at ./.obj-x86_64-linux-gnu/gz_ogre_next_vendor-prefix/src/gz_ogre_next_vendor/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 165)
[GUI] [Err] [Ogre2RenderEngine.cc:1301]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at ./.obj-x86_64-linux-gnu/gz_ogre_next_vendor-prefix/src/gz_ogre_next_vendor/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 165)
[GUI] [Err] [Ogre2RenderEngine.cc:1301]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at ./.obj-x86_64-linux-gnu/gz_ogre_next_vendor-prefix/src/gz_ogre_next_vendor/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 165)
[GUI] [Err] [Ogre2RenderEngine.cc:1301]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at ./.obj-x86_64-linux-gnu/gz_ogre_next_vendor-prefix/src/gz_ogre_next_vendor/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 165)
[GUI] [Err] [Ogre2RenderEngine.cc:1309] Unable to create the rendering window after [11] attempts.
[GUI] [Err] [Ogre2RenderEngine.cc:1191] Failed to create dummy render window.
[GUI] [Err] [Ogre2RenderEngine.cc:1192] Please see the troubleshooting page for possible fixes: https://gazebosim.org/docs/fortress/troubleshooting
Stack trace (most recent call last):
#31   Object "/opt/ros/jazzy/opt/gz_sim_vendor/lib/libgz-sim8-gui.so.8", at 0x7217754755f0, in gz::sim::v8::gui::runGui(int&, char**, char const*, char const*, int, char const*, char const*)
#30   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7217740df3e7, in QCoreApplication::exec()
#29   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7217740d6a7a, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#28   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x721774135278, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#27   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x721775313a52, in g_main_context_iteration
#26   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x721775373716, in 
#25   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7217753145b4, in 
#24   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x721774135c0e, in 
#23   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7217740db94a, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
#22   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7217740d8117, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#21   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x72177376bd44, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#20   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x721774106342, in QObject::event(QEvent*)
[... cut ...] string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::STLAllocator<std::pair<Ogre::IdString, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
Segmentation fault (Address not mapped to object [0x220])
```