gazebosim/gz-sim #2362: Make child model and link configurable in DetachableJoint System

jasmeet0915, 2024-04-05:
## Desired behavior
Currently, the detachable joint system allows you to attach or detach a joint between a parent and a child model by publishing an Empty message on the respective topics. For this to work we have to specify the child model and link name in the SDF of the plugin. However, while simulating some real world operations I found the need of allowing my simulated robot to be able to create 'detachable joints' with different child models at different times. 

This is similar functionality like the [`gazebo_ros_link_attacher`](https://github.com/pal-robotics/gazebo_ros_link_attacher/tree/melodic-devel) plugin which provided a service to attach and detach joints between any two models as per the request.

## Implementation suggestion
Currently the `detachable_joint` system assigns the [modelEntity](https://github.com/gazebosim/gz-sim/blob/b9138fb12b0cef77621ed3a5eba577ebe4f7abd0/src/systems/detachable_joint/DetachableJoint.cc#L237) and [childLinkEntity](https://github.com/gazebosim/gz-sim/blob/b9138fb12b0cef77621ed3a5eba577ebe4f7abd0/src/systems/detachable_joint/DetachableJoint.cc#L242) in the `PreUpdate()` function of the plugin using the `childModelName` and `childLinkName` members that are set in the [`Configure()`](https://github.com/gazebosim/gz-sim/blob/b9138fb12b0cef77621ed3a5eba577ebe4f7abd0/src/systems/detachable_joint/DetachableJoint.cc#L76) function from the values given in the SDF. So I think instead of having a separate attach / detach topic with Empty msg type, it should be possible to advertise a service with a custom message type of detachable joint msg composed of string fields that allows you to set the `child_model_name`, `child_link_name` and `operation` (attach or detach). In the callback of the service, we could set the requested values for the required data members. The detachable_joint service could also be added to the [`ros_gz_interfaces`](https://github.com/gazebosim/ros_gz/tree/ros2/ros_gz_interfaces) package to allow bridging of this service.

I would be happy to create a pull request implementing the changes mentioned if they sound right!

AdarshKaran, 2025-09-03:
Thank you for the feedback! I totally get it- no rush to merge. I kept this as a separate plugin because I didn’t want to break the current `detachable_joint` behavior. The new `dynamic_detachable_joint` reuses the existing plugin but makes it more flexible.

I’m happy to try adding these into the existing detachable joint plugin if that’s preferred. I could open PRs across the repos (`gz-msgs`, `gz-sim`, `ros_gz_interfaces`, `ros_gz_bridge`) so you can see the diffs and suggest improvements. Or please let me know what we can do. Maybe after taking a look at the diffs we could decide whether to integrate into the current plugin or keep it separate if that's fine. Let me know which direction you’d like me to take.

arjo129, 2025-09-03:
Open the PR first, we can iterate and discuss over time.

AdarshKaran, 2025-09-03:
Hi @arjo129 I have a small query regarding the PRs. It's been a while since i made these changes and there are newer updates to the default branches of the above pkgs. Shall I pull the changes, merge and then give PR?

arjo129, 2025-09-03:
It doesn't matter. I'd recommend merging the opening, but either way doesn't matter. Try to target `main`.

AdarshKaran, 2025-09-03:
Thanks for the clarification @arjo129. I'll give the PR's now maybe once the changes you suggest are finalized I could make sure that it merges with the main branch without conflicts.

Divyam-uga, 2025-09-10:
Hi @AdarshKaran,

I noticed you had a working version related to this issue. I’m currently working on a project that involves 4 robots where each robot needs to dynamically select and pick up an onion. Would you be able to share your code or the pull request where you implemented this? It would be a huge help for my research setup.

AdarshKaran, 2025-10-21:
> Hi [@AdarshKaran](https://github.com/AdarshKaran),
> 
> I noticed you have a working solution for this issue. I’m currently working on a project that involves 4 robots where each robot needs to dynamically select and pick up an onion. Would you be able to share your code or PR? It would be a huge help for my research setup.

Hi @Divyam-uga ! Just curious if you had a chance to test the solution yet? If so I would love to get your feedback on how it went or if it addressed the issue for you.

AdarshKaran, 2025-10-21:
> Thank you [@AdarshKaran](https://github.com/AdarshKaran)! I’m glad you liked the setup.
> 
> You’re absolutely welcome to include clips from my Gazebo sim videos in your ROSCon 2025 presentation.
> 
> Wishing you all the best with your lightning talk submission!

thank you @Divyam-uga !

closed 2026-11-06 by iche033