gazebosim/gz-sim #2364: ../src/intel/isl/isl.c:2105: FINISHME: ../src/intel/isl/isl.c:isl_surf_supports_ccs: CCS for 3D textures is disabled, but a workaround is available.

AnishMathewOommen, 2024-04-07:
Please help me to solve this issue

Using Ros : Noetic (20.04 version)
When I run gazebo I get the below error
gazebo
../src/intel/isl/isl.c:2105: FINISHME: ../src/intel/isl/isl.c:isl_surf_supports_ccs: CCS for 3D textures is disabled, but a workaround is available.

azeey, 2024-04-08:
If you think this is a bug, please fill out the issue [template](https://github.com/gazebosim/gz-sim/blob/gz-sim8/.github/ISSUE_TEMPLATE/bug_report.md) provided when you opened the issue. If you think it might be an issue with your local setup, https://robotics.stackexchange.com/ would be a better place to ask.

AnishMathewOommen, 2024-04-12:
Thank you for your help @azeey

AnishMathewOommen, 2024-04-15:
**Found the Solution for the above problem**
Please type the below commands in the terminal the issue will be solved( Updating Mesa driver)

```
sudo add-apt-repository ppa:kisak/kisak-mesa
sudo apt update
sudo apt upgrade

```

Reference document: https://itsfoss.com/install-mesa-ubuntu/

closed 2024-04-15 by AnishMathewOommen