gazebosim/gz-sim #2366: Wrong URI for Worlds in StartupGUI

drewskoots, 2024-04-10:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build?
 Binary Build, version 8.2.0

This seems similar to #2281 seeing as the URI that is trying to be downloaded starts with fuel.ignitionrobotics.org. Downloading worlds directly from the Fuel website seems to work fine.

## Description
* Expected behavior: Upon selecting an example world, it is downloaded from fuel and started.
* Actual behavior: The world download receives a 504 error and the simulation hangs/crashes

## Steps to reproduce

1. clean install gz harmonic as described in the docs
2. run `gz sim`
3. try to open any of the 5 example worlds shown in the dialog box that opens (other than empty of course).

## Output
```
Err] [FuelClient.cc:695] Failed to download model.
  Server: https://fuel.ignitionrobotics.org
  Route: openrobotics/models/panda with ignition position controller model/tip/panda with ignition position controller model.zip
  REST response code: 504
[Err] [FuelClient.cc:695] Failed to download model.
  Server: https://fuel.ignitionrobotics.org
  Route: openrobotics/models/table/tip/table.zip
  REST response code: 504
[Err] [FuelClient.cc:695] Failed to download model.
  Server: https://fuel.ignitionrobotics.org
  Route: openrobotics/models/table/tip/table.zip
  REST response code: 504
```

drewskoots, 2024-04-10:
Additionally, I've notice the resource spawner also fails silently (unless -v 4 is selected)
```
[GUI] [Dbg] [Application.cc:528] Loading plugin [ResourceSpawner]
[GUI] [Wrn] [ResourceSpawner.cc:614] No paths found in GZ_SIM_RESOURCE_PATH.
[GUI] [Msg] Please wait... Loading models from Fuel.
[GUI] [Msg] fuel.gazebosim.org/openrobotics/models/ not found in cache, attempting download
[GUI] [Msg] Added plugin [Resource Spawner] to main window
[GUI] [Msg] Loaded plugin [ResourceSpawner] from path [/usr/lib/x86_64-linux-gnu/gz-sim-8/plugins/gui/libResourceSpawner.so]
[GUI] [Wrn] [ResourceSpawner.cc:589] Download failed.  Try again.
[GUI] [Wrn] [ResourceSpawner.cc:589] Download failed.  Try again.

```
I was not able to determine if this is also because of wrong URI

azeey, 2024-05-06:
There was an infrastructure issue with fuel.gazebosim.org that was causing these problems. Please feel free to open if you're still having these issues.

closed 2024-05-06 by azeey

dp1140a, 2024-05-24:
I am also encountering this error on harmonic whenever I try to download a model form the Resource Spawner