gazebosim/gz-sim #2372: DiffDrive publish_odom_tf parameter

Wiktor-99, 2024-04-15:
## Desired behavior
In the [gazebo_ros_diff_drive](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/ros2/gazebo_plugins/src/gazebo_ros_diff_drive.cpp) publishing wheels TF and "odom" => "robot_base_frame" TF is split.
 
Is it even possible to implement such behavior in the Ignition Gazebo? If there is feasible solution I could work on that.

## Additional context
There is problem when you want to use robot_localization pkg which also provides TF to  "odom" => "robot_base_frame", propably most of the time users would like to use TF provided by robot_localization pkg instead of DiffDrive plugin.

azeey, 2024-04-16:
The `DiffDrive` plugin currently doesn't publish wheels TF. For that you need the `JointStatePublisher` plugin and `robot_state_publisher` on the ROS side. This also means, if you want to provide your own `odom -> robot_base_frame`, you simply have to ignore the TF messages published by DiffDrive (by not bridging the topic).

Wiktor-99, 2024-04-17:
Yes, you are right my assumptions were wrong, issue could be closed. Nevertheless when `JointStatePublisher` and `DiffDrive` are used EKF node dose not work with `use_sim_time` param. Probably I messed up configuration, fortunately ros2_control worked for me.

closed 2024-04-17 by Wiktor-99