gazebosim/gz-sim #2376: gz sim specific.sdf - run random world if present more than 1

drfenixion, 2024-04-18:
## Environment
* OS Version: Ubuntu 22.04.4 LTS

binary Gazebo Garden 7.7.0

## Description
* Expected behavior: 
I run  `gz sim` (for making any other word). In other terminal i run  `gz sim fly.sdf`. I want to see second Gazebo window as fly.sdf but getting random world (default or fly.sdf).

* Actual behavior: 
Getting random world (default or specific.sdf) in second window of Gazebo. It is unstable cab be many times (1-10) default.sdf or specific.sdf.

If world only 1 (in gz service --list) all works fine.

## Steps to reproduce
1.  I run  `gz sim` 
2.  In other terminal i run  `gz sim specific.sdf`. Try to do 10 times for get statistics of results (mean run than see result and ctrl + c, after try again).

## Output
`gz service --list`
/gazebo/resource_paths/add
/gazebo/resource_paths/get
/gazebo/resource_paths/resolve
/gazebo/worlds
/gui/camera/view_control
/gui/camera/view_control/reference_visual
/gui/camera/view_control/sensitivity
/gui/copy
/gui/follow
/gui/follow/offset
/gui/move_to
/gui/move_to/pose
/gui/paste
/gui/screenshot
/gui/view/collisions
/gui/view/com
/gui/view/frames
/gui/view/inertia
/gui/view/joints
/gui/view/transparent
/gui/view/wireframes
/marker
/marker/list
/marker_array
/server_control
/world/default/control
/world/default/control/state
/world/default/create
/world/default/create_multiple
/world/default/declare_parameter
/world/default/disable_collision
/world/default/enable_collision
/world/default/entity/system/add
/world/default/generate_world_sdf
/world/default/get_parameter
/world/default/gui/info
/world/default/level/set_performer
/world/default/light_config
/world/default/list_parameters
/world/default/playback/control
/world/default/remove
/world/default/scene/graph
/world/default/scene/info
/world/default/set_parameter
/world/default/set_physics
/world/default/set_pose
/world/default/set_pose_vector
/world/default/set_spherical_coordinates
/world/default/state
/world/default/state_async
/world/default/system/info
/world/default/visual_config
/world/default/wheel_slip
/world/fly/control
/world/fly/control/state
/world/fly/create
/world/fly/create_multiple
/world/fly/declare_parameter
/world/fly/disable_collision
/world/fly/enable_collision
/world/fly/entity/system/add
/world/fly/generate_world_sdf
/world/fly/get_parameter
/world/fly/gui/info
/world/fly/level/set_performer
/world/fly/light_config
/world/fly/list_parameters
/world/fly/playback/control
/world/fly/remove
/world/fly/scene/graph
/world/fly/scene/info
/world/fly/set_parameter
/world/fly/set_physics
/world/fly/set_pose
/world/fly/set_pose_vector
/world/fly/set_spherical_coordinates
/world/fly/state
/world/fly/state_async
/world/fly/system/info
/world/fly/visual_config
/world/fly/wheel_slip

Screen of worlds:
default
[image]
fly.sdf
[image]
:

azeey, 2024-04-18:
It's not clear to me what you're trying to do. Are you trying to run two different instances of Gazebo with two different worlds at the same time?

drfenixion, 2024-04-18:
It is synthetic case only for bug demonstration.

I discovered the bug by accident. I worked a lot with the launcher and launched different Gazebo maps. Somehow one of them did not unload(although it should have unloaded) from worlds and for a long time I could not understand why a random map began to open instead of the desired one.

azeey, 2024-05-06:
So, the issue is more about accidentally running two instances of Gazebo because, in your example, the `ros_gz` launcher did not exit properly?

drfenixion, 2024-05-06:
I believe it is about unpredictable behaviour (bug) of Gazebo. Because if i run gz sim specific.sdf i expect it should runs it regardless of what worlds were loaded before.

arjo129, 2024-06-14:
I think I see the problem here. The client GUI may not pick the right world.