gazebosim/gz-sim #2377: Calling service for getting Actor poses

Rak-r, 2024-04-19:
Version: Gazebo Fortress

I am trying to call the service to get the poses of actors in the simulation from command line using

` ign service -s /world/actors/state --reqtype ignition.msgs.Actor --reptype ignition.msgs.Actor --timeout -50000 --req 'entity: {name: "demo_actor"}, pose: {}'`

But nothing comes up in the terminal after running the above. Am I missing something or is the syntex is wrong? Any help would be much appreciated. @azeey 

**Requirement:**  

I want the pose as topic so that I can communicate with ros2.
Is there any way to call this service through python script? any sample example is available. I checked the sim repo but not able to find much on this.

azeey, 2024-04-19:
Have you tried the `PosePublisher` system with `<publish_nested_model_pose>` set to true?

Rak-r, 2024-04-19:
I believe that works fro models entities not actor. I have tried adding it but not running as it asks for model to specified under model tag not actor.

Riyadh-kh, 2024-04-23:
Referencing issue #2370 the same problem is occurring with me, I can't find a way to access actors poses despite using `PosePublisher`

Rak-r, 2024-04-23:
I was thinking to implement some custom plugin to publish both cmd_vel and pose from ros2 and control the actor. Something similar exists for ros2 and gazebo 11 but new headers and classes in ignition gazebo or whatever the version of new gazebo  it is a bit confusing. I am trying by taking reference from commamd actor plugin.@Riyadh-kh

azeey, 2024-05-13:
Can we close this as a duplicate of #2370 ?

azeey, 2024-07-15:
Because of the extra context in this ticket, let's keep this is the canonical ticket and close #2370. Any help with the implementation would be greatly appreciated.