gazebosim/gz-sim #2380: Docs: Add tutorial on how to start/pause simulator from CLI

Ryanf55, 2024-04-23:
## Desired behavior
Add  a new tutorial similar to [reset](https://github.com/gazebosim/gz-sim/blob/gz-sim8/tutorials/reset_simulation.md) that shows how to start/stop. 

## Alternatives considered
N/A

## Implementation suggestion

* Show start/pause button in GUI and what it does
* Explain how sim time stops and how to check sim time and iterations in CLI
* Explain how to start/pause with the `gz` CLI through transport
* Add a demo video

```bash
gz sim -s shapes.sdf
gz service -s /world/shapes/control --reqtype gz.msgs.WorldControl --reptype gz.msgs.Boolean --timeout 3000 --req 'pause: false'
gz topic -e -t /stats -n 1
sim_time {
  sec: 2
  nsec: 807000000
}
real_time {
  sec: 2
  nsec: 824597858
}
iterations: 2807
real_time_factor: 0.99667211181863757
step_size {
  nsec: 1000000
}

## Additional context

Please assign this to me. I'll write the tutorial.

Ryanf55, 2024-04-24:
I would like to create a `gif` like the other tutorial. Can the maintainers recommend a process to do that? Or, is it OK to do the tutorial without a gif. Otherwise, I have the content ready.

mjcarroll, 2024-04-30:
> I would like to create a gif like the other tutorial. Can the maintainers recommend a process to do that? Or, is it OK to do the tutorial without a gif. Otherwise, I have the content ready.

I don't know that we have a guide, but we have historically used https://en.wikipedia.org/wiki/Peek_(software) to record gifs on Linux.

Ryanf55, 2024-04-30:
Thanks. Which world is shown in other tutorial?

PR #2383 (2024-06-11): Add pause run tutorial
#  📚 Docs

Fixes #2380

## Summary

Add a tutorial to start/stop the simulation with either the GUI or CLI. This mirrors the reset simulation tutorial.

## Checklist

- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://gazebosim.org/docs/all/contributing#test-coverage) [... cut]
closed 2024-06-11 by azeey