gazebosim/gz-sim #2384: :farmer: sim6 windows CI: 13 warnings

Crola1702, 2024-04-25:
## Environment
* OS Version: Windows
* Source gz-sim6 build

## Description
There are 13 warnings in [gz_sim-ign-gazebo6-win](https://build.osrfoundation.org/job/gz_sim-ign-gazebo6-win/42/)

## Steps to reproduce
1. Run a build in [gz_sim-ign-gazebo6-win](https://build.osrfoundation.org/job/gz_sim-ign-gazebo6-win/42/)
2. See warnings appear

## Output

Reference build: https://build.osrfoundation.org/job/gz_sim-ign-gazebo6-win/42/

Warnings (See attached hyperlinks for more details):
* 9 [C4273](https://build.osrfoundation.org/job/gz_sim-ign-gazebo6-win/42/msbuild/category.63474845/): inconsistent dll linkage
* 1 [C4834](https://build.osrfoundation.org/job/gz_sim-ign-gazebo6-win/42/msbuild/category.63480488/): discarding return value of function with 'nodiscard' attribute
* 3 [C4584](https://build.osrfoundation.org/job/gz_sim-ign-gazebo6-win/42/msbuild/category.63477760/): 'ignition::physics::detail::ComposePlugin::type': base-class 'ignition::physics::detail::ComposePlugin::n,void>' is already a base-class of 'ignition::physics::detail::ComposePlugin::type'

Blast545, 2024-06-03:
:farmer: I think this could be solved by a combination of https://github.com/gazebosim/gz-sim/pull/1911/ and https://github.com/gazebosim/gz-sim/pull/2254 being backported to sim6. 

@mjcarroll Can I ask you to take this one? I saw you opened the two mentioned PRs, so you may have more context if anything goes wrong with those. :pray:

claraberendsen, 2024-06-17:
friendly ping @mjcarroll

claraberendsen, 2024-07-15:
Most of the warnings seem to have been addressed. 
However 
- 3 [C4584](https://build.osrfoundation.org/job/gz_sim-ign-gazebo6-win/42/msbuild/category.63477760/): 'ignition::physics::detail::ComposePlugin::type': base-class 'ignition::physics::detail::ComposePlugin::n,void>' is already a base-class of 'ignition::physics::detail::ComposePlugin::type'

keeps happening. 
See : https://build.osrfoundation.org/job/gz_sim-ign-gazebo6-win/68/ that ran on Jul 12th.

Blast545, 2024-07-16:
I took some time to dig a bit further, the code complaining in the warning is not in the gazebo repo, it's in the gz-physics one.

In gz-physics7 that warning is disabled: https://github.com/gazebosim/gz-physics/blob/cca7e923702b1711f4f16db8c869eec5e710acf7/include/gz/physics/FeatureList.hh#L25

So I think it's just a matter of backporting https://github.com/gazebosim/gz-physics/commit/f2775ac00fd76ca46316ac3d1bbdc1cdb9cc00fa to physics5, can you take a look @Crola1702?

Crola1702, 2024-07-16:
The backport is already on the way. See: https://github.com/gazebosim/gz-physics/pull/666

Blast545, 2024-07-19:
Sim6 has been green in the last two builds, I'm closing this one, thanks a lot @mjcarroll @Crola1702 @azeey and @claraberendsen!

closed 2024-07-19 by Blast545