gazebosim/gz-sim #2390: Binary installation on Ubuntu 24.04 doesn't seem to work

suren-atoyan, 2024-04-28:
## Environment
* OS Version: Ubuntu 24.04
* Source or binary build? Binary, Harmonic

## Description

I used the installation steps described [here](https://gazebosim.org/docs/harmonic/install_ubuntu). Step `sudo apt-get install gz-harmonic` gives `E: Unable to locate package gz-harmonic`.

* Expected behavior: find and install `gz-harmonic` on Ubuntu 24.04
* Actual behavior: Unable to locate package

## Steps to reproduce

1. Fresh install of Ubuntu 24.04 and ROS2 Jazzy
2. Installation steps of gazebo from [here](https://gazebosim.org/docs/harmonic/install_ubuntu)
3. `sudo apt install gz-harmonic`

azeey, 2024-05-06:
Ubuntu 24.04 just came out so we haven't built the binaries for it yet on packages.osrfoundation.org. It should be available in the coming few weeks. cc @j-rivero 

If you're installing Jazzy, you can just install `ros-jazzy-ros-gz` and it will install Gazebo Harmonic from packages.ros.org via the gz vendor packages (documentation to come soon).

hikmetkara, 2024-05-10:
Does binaries coming soon? We are still waiting sir.

mjcarroll, 2024-05-14:
> Does binaries coming soon? We are still waiting sir.

You can install the binaries from the ROS buildfarm as indicated above

`sudo apt install ros-jazzy-ros-gz`

j-rivero, 2024-05-14:
https://community.gazebosim.org/t/gazebo-harmonic-packages-available-on-ubuntu-noble/2772

closed 2024-05-14 by j-rivero

dp1140a, 2024-05-25:
@mjcarroll Where is the ROS buildfarm?

I tried:
```bash
sudo sh -c 'echo "deb [arch=$(dpkg --print-architecture)] http://packages.ros.org/ros2/ubuntu $(lsb_release -cs) main" > /etc/apt/sources.list.d/ros2-latest.list'
curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | sudo apt-key add -
sudo apt-get update
sudo apt install ros-jazzy-ros-gz
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-jazzy-ros-gz
```