gazebosim/gz-sim #2399: gz topic not being published...

Marco-63, 2024-05-08:
## Environment
* OS Version: Ubuntu 22.04 running in a Oracle Virtual Box 7.0 hosted on Linux Mint 
* Source or binary build?
  Same behavior building from source or binary - latest Harmonic

## Description
* Expected behavior: Robot starts moving
* Actual behavior: Absolutely nothing happens!

## Steps to reproduce
1. In a terminal run: **gz sim -r building_robot.sdf**
2. In another terminal run: **gz topic -t "/cmd_vel" -m gz.msgs.Twist -p "linear: {x: 0.5}, angular: {z: 0.05}"**
3. Now you should have your robot moving in the simulation...!

## Comment
* Topic command never reaches the simulation world and therefore the robot doesn't start to move...

azeey, 2024-05-08:
The topic `/cmd_vel` is configured by the system in your SDF file. Have you set it up properly? Assuming you were following the https://gazebosim.org/docs/harmonic/moving_robot tutorial, the final version of the SDF file is available [here](https://github.com/gazebosim/docs/blob/master/harmonic/tutorials/moving_robot/moving_robot.sdf). Can you try with that?

Marco-63, 2024-05-09:
Hi Azeey,
thank you for your reply!
I confirm that my SDF file is exactly the same as the one you linked and unfortunately, of course, the problem is the same as described above: the robot does not move...

This is the command I tried:
`$ gz topic -t "/cmd_vel" -m gz.msgs.Twist -p "linear: {x: 0.5}, angular: {z: 0.05}"`

and this is the topic --info:
```
$ gz topic -t /cmd_vel --info
Publishers [Address, Message Type]:
  tcp://10.0.2.15:36663, gz.msgs.Twist
  tcp://10.0.2.15:36663, gz.msgs.Twist
  tcp://10.0.2.15:36663, gz.msgs.Twist
  tcp://10.0.2.15:36663, gz.msgs.Twist
Subscribers [Address, Message Type]:
  tcp://10.0.2.15:36663, gz.msgs.Twist

```
Pls. let me know what you need to see... thx!

azeey, 2024-05-09:
It works for me:
[Screen recording 2024-05-09 11.16.32.webm](https://github.com/gazebosim/gz-sim/assets/206116/b12c8db5-9c59-4673-8678-23518a297509)

Are you able to do `gz topic -e -t /clock`? You might have an issue with your firewall if that doesn't work.

Marco-63, 2024-05-09:
The clock topic reads fine... here is a sample:
```
system {
  sec: 1715272967
  nsec: 710552530
}
real {
  sec: 276
  nsec: 608780872
}
sim {
  sec: 172
  nsec: 966000000
}
```
My firewall says: "status: inactive"  with "sudo ufw status"
Sorry, unfortunately only /cmd_vel doesn't seem to work...

Any other suggestions? thx!

azeey, 2024-05-09:
Can you run the simulation with `-v4` to see if there are any errors?

Marco-63, 2024-05-09:
Pls. take a look at this, lots of warnings but doesn't seem to be any errors...

<pre>$ gz sim -r -v4  building_robot.sdf 
<font color="#2AA1B3"><b>[Dbg] [gz.cc:166] Subscribing to [/gazebo/starting_world].</b></font>
<font color="#26A269"><b>[Msg] Gazebo Sim GUI    v8.3.0</b></font>
<font color="#2AA1B3"><b>[Dbg] [gz.cc:168] Waiting for a world to be set from the GUI...</b></font>
<font color="#2AA1B3"><b>[Dbg] [Gui.cc:263] Waiting for subscribers to [/gazebo/starting_world]...</b></font>
<font color="#26A269"><b>[Msg] Received world [building_robot.sdf] from the GUI.</b></font>
<font color="#2AA1B3"><b>[Dbg] [gz.cc:172] Unsubscribing from [/gazebo/starting_world].</b></font>
<font color="#26A269"><b>[Msg] Gazebo Sim Server v8.3.0</b></font>
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
<font color="#26A269"><b>[Msg] Loading SDF world file[/home/monti/Downloads/GazeboTutorial/building_robot.sdf].</b></font>
<font color="#26A269"><b>[Msg] Serving entity system service on [/entity/system/add]</b></font>
<font color="#2AA1B3"><b>[Dbg] [Physics.cc:860] Loaded [gz::physics::dartsim::Plugin] from library [/home/monti/Downloads/GazeboHarmonic/install/lib/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so]</b></font>
<font color="#2AA1B3"><b>[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Physics] for entity [1]</b></font>
<font color="#2AA1B3"><b>[Dbg] [Application.cc:96] Initializing application.</b></font>
<font color="#2AA1B3"><b>[Dbg] [Application.cc:170] Qt using OpenGL graphics interface</b></font>
<font color="#2AA1B3"><b>[GUI] [Dbg] [Application.cc:657] Create main window</b></font>
<font color="#26A269"><b>[Msg] Create service on [/world/car_world/create]</b></font>
<font color="#26A269"><b>[Msg] Remove service on [/world/car_world/remove]</b></font>
<font color="#26A269"><b>[Msg] Pose service on [/world/car_world/set_pose]</b></font>
<font color="#26A269"><b>[Msg] Pose service on [/world/car_world/set_pose_vector]</b></font>
<font color="#26A269"><b>[Msg] Light configuration service on [/world/car_world/light_config]</b></font>
<font color="#26A269"><b>[Msg] Physics service on [/world/car_world/set_physics]</b></font>
<font color="#26A269"><b>[Msg] SphericalCoordinates service on [/world/car_world/set_spherical_coordinates]</b></font>
<font color="#26A269"><b>[Msg] Enable collision service on [/world/car_world/enable_collision]</b></font>
<font color="#26A269"><b>[Msg] Disable collision service on [/world/car_world/disable_collision]</b></font>
<font color="#26A269"><b>[Msg] Material service on [/world/car_world/visual_config]</b></font>
<font color="#26A269"><b>[Msg] Material service on [/world/car_world/wheel_slip]</b></font>
<font color="#2AA1B3"><b>[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::UserCommands] for entity [1]</b></font>
<font color="#2AA1B3"><b>[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::SceneBroadcaster] for entity [1]</b></font>
<font color="#2AA1B3"><b>[Dbg] [TriggeredPublisher.cc:689] TriggeredPublisher subscribed on /keyboard/keypress and publishing on /cmd_vel, </b></font>
<font color="#2AA1B3"><b>[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::TriggeredPublisher] for entity [1]</b></font>
<font color="#2AA1B3"><b>[Dbg] [TriggeredPublisher.cc:689] TriggeredPublisher subscribed on /keyboard/keypress and publishing on /cmd_vel, </b></font>
<font color="#2AA1B3"><b>[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::TriggeredPublisher] for entity [1]</b></font>
<font color="#2AA1B3"><b>[Dbg] [TriggeredPublisher.cc:689] TriggeredPublisher subscribed on /keyboard/keypress and publishing on /cmd_vel, </b></font>
<font color="#2AA1B3"><b>[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::TriggeredPublisher] for entity [1]</b></font>
<font color="#2AA1B3"><b>[Dbg] [TriggeredPublisher.cc:689] TriggeredPublisher subscribed on /keyboard/keypress and publishing on /cmd_vel, </b></font>
<font color="#2AA1B3"><b>[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::TriggeredPublisher] for entity [1]</b></font>
<font color="#26A269"><b>[Msg] DiffDrive subscribing to twist messages on [cmd_vel]</b></font>
<font color="#2AA1B3"><b>[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::DiffDrive] for entity [8]</b></font>
<font color="#26A269"><b>[Msg] Loaded level [3]</b></font>
<font color="#26A269"><b>[Msg] Serving world controls on [/world/car_world/control], [/world/car_world/control/state] and [/world/car_world/playback/control]</b></font>
<font color="#26A269"><b>[Msg] Serving GUI information on [/world/car_world/gui/info]</b></font>
<font color="#26A269"><b>[Msg] World [car_world] initialized with [1ms] physics profile.</b></font>
<font color="#26A269"><b>[Msg] Serving world SDF generation service on [/world/car_world/generate_world_sdf]</b></font>
<font color="#26A269"><b>[Msg] Serving world names on [/gazebo/worlds]</b></font>
<font color="#26A269"><b>[Msg] Resource path add service on [/gazebo/resource_paths/add].</b></font>
<font color="#26A269"><b>[Msg] Resource path get service on [/gazebo/resource_paths/get].</b></font>
<font color="#26A269"><b>[Msg] Resource path resolve service on [/gazebo/resource_paths/resolve].</b></font>
<font color="#26A269"><b>[Msg] Resource paths published on [/gazebo/resource_paths].</b></font>
<font color="#26A269"><b>[Msg] Server control service on [/server_control].</b></font>
[... cut ...]
<font color="#26A269"><b>[GUI] [Msg] Move to pose service on [/gui/move_to/pose]</b></font>
<font color="#26A269"><b>[GUI] [Msg] Camera pose topic advertised on [/gui/camera/pose]</b></font>
<font color="#26A269"><b>[GUI] [Msg] Follow offset service on [/gui/follow/offset]</b></font>
<font color="#2AA1B3"><b>[GUI] [Dbg] [InteractiveViewControl.cc:176] InteractiveViewControl plugin is moving camera [scene::Camera(65527)]</b></font>
<font color="#2AA1B3"><b>[GUI] [Dbg] [EntityContextMenuPlugin.cc:79] Entity context menu plugin is using camera [scene::Camera(65527)]</b></font>

</pre>

Marco-63, 2024-05-17:
Hello Azeey,
please let me know if it would be it ok if I post a link from WeTransfer where you can download my Ubuntu VirtualBox machine so that maybe you can see what's going wrong more easily?

Thank you,
Marco

azeey, 2024-05-20:
Unfortunately, I don't have enough cycles to diagnose the problem that way. I noticed from the logs you have built Gazebo from source. Have you tried the other examples in the `examples/worlds` directory? Does `rolling_shapes.sdf` work properly where the shapes roll down the incline?

Marco-63, 2024-05-23:
Yes, I built Gazebo from source (5 hrs job!) because a had found a post suggesting that it might solve the problem. Unfortunately, noting changes... The problem remains the same: topic /cmd_vel doesn't get published!

I tried to run rolling_shapes.sdf and it works fine!

I would like to show you the following... if you do this:
```
$ gz topic -t /cmd_vel --info
Publishers [Address, Message Type]:
  tcp://10.0.2.15:34523, gz.msgs.Twist
  tcp://10.0.2.15:34523, gz.msgs.Twist
  tcp://10.0.2.15:34523, gz.msgs.Twist
  tcp://10.0.2.15:34523, gz.msgs.Twist
Subscribers [Address, Message Type]:
  tcp://10.0.2.15:34523, gz.msgs.Twist

$ sudo netstat -tulpn | grep LISTEN
tcp        0      0 10.0.2.15:34871         0.0.0.0:*               LISTEN      12125/gz sim gui    
tcp        0      0 10.0.2.15:46449         0.0.0.0:*               LISTEN      12124/gz sim server 
tcp        0      0 10.0.2.15:41705         0.0.0.0:*               LISTEN      12125/gz sim gui    
tcp        0      0 10.0.2.15:34523         0.0.0.0:*               LISTEN      12124/gz sim server 
tcp        0      0 10.0.2.15:45757         0.0.0.0:*               LISTEN      12124/gz sim server 
tcp        0      0 10.0.2.15:44675         0.0.0.0:*               LISTEN      12125/gz sim gui    
tcp        0      0 127.0.0.1:631           0.0.0.0:*               LISTEN      10009/cupsd         
tcp        0      0 127.0.0.53:53           0.0.0.0:*               LISTEN      382/systemd-resolve 
tcp6       0      0 ::1:631                 :::*                    LISTEN      10009/cupsd         

```
the TCP ports are open but no activity seems to be going on through these ports, if you look at network traffic with Wireshark.
First time I use Wireshark, but it seems clear...

[image]

this is the only TCP activity in the VM...

Does this mean anything to you?
Thx!

azeey, 2024-05-23:
I wonder if you have multiple versions of `gz-transport` installed. Can you try running `gz topic --versions`?
You can also set the environment variable `GZ_VERBOSE=1` before running any `gz` command it will show more debug messages related to transport.

Marco-63, 2024-05-27:
Hi Azeey,
running gz topic --version gives 13.2.0
After setting the environment variable GZ_VERBOSE I tried to simulate building_robot.sdf. There was a lot of debug messages coming out, but unfortunately I can't figure much out of it. So I'am attaching two files with the logs: one file contains the log after issuing the topic command and the other what came out in the gazebo terminal...
If you can spare 5 more minutes, please have a look at them. Thx!

[cmdTopic.txt](https://github.com/gazebosim/gz-sim/files/15456737/cmdTopic.txt)
[Gazebo_after.txt](https://github.com/gazebosim/gz-sim/files/15456748/Gazebo_after.txt)

azeey, 2024-05-30:
That looks okay to me. Can you share the `building_robot.sdf` file you are using?

Marco-63, 2024-05-31:
I'm starting to think that Gazebo is not able to run in a VirtualBox VM. Do you believe this could be the origin of the problem?

Here is my .sdf file: [building_robot.txt](https://github.com/user-attachments/files/15511070/building_robot.txt) (I just changed the extendion to .txt to be able to load the file)

Thx!

azeey, 2024-05-31:
Yeah, I'm suspecting the same thing. It's surprising though since usually the problem is 3D acceleration, not gz-transport communication. I tried your file and it works fine for me locally.

I'm curious why you're using a VM. If your host is Linux Mint, you should be able to use docker. Have you considered that?

Marco-63, 2024-06-04:
Hello Azeey,
at the end of this long story, the reason it didn't work was the virtual machine. I installed the binaries in a physical computer and gz-transport started to communicate!
I'm used to try new things in a VM first and this is why I did it:..

Sorry for all the troubles and  thank you very very much for your support.
Now I can finally start experimenting with Gazebo!

azeey, 2024-06-04:
I suspect there's some configuration that needs to be set on the VM or host machine to allow gz-transport to work. I've seen others use VirtualBox with Gazebo, but this is the first time I'm hearing of such problems, so I don't want to conclude Gazebo doesn't work on VirtualBox. That being said, I'm glad to hear it's working for you on the host machine. I'll go ahead and close this issue.

closed 2024-06-04 by azeey