gazebosim/gz-sim #2400: worldPose() helper function in Util.hh does not properly compute world poses for Joints

scpeters, 2024-05-08:
## Description
* Expected behavior: the [`worldPose()` helper function in Util.hh](https://github.com/gazebosim/gz-sim/blob/gz-sim8_8.3.0/include/gz/sim/Util.hh#L46-L51) should be able to compute the world pose of a Joint.
* Actual behavior: it does not work properly (see [note in ForceTorque system](https://github.com/gazebosim/gz-sim/blob/gz-sim8_8.3.0/src/systems/force_torque/ForceTorque.cc#L290-L295))

I believe this is because the recursive pose calculation computes the joint's pose relative to its parent model, rather than relative to its child link ([as defined in SDFormat](http://sdformat.org/tutorials?tut=pose_frame_semantics&ver=1.7&cat=specification&#implicit-and-explicit-frames)).

## Steps to reproduce

1. TODO: update Unit test to demonstrate the error
2.
3.

## Output

azeey, 2024-05-08:
Related to https://github.com/gazebosim/gz-sim/issues/1042#issue-1004542406

scpeters, 2024-05-13:
ok, I'll move discussion to #1042

scpeters, 2024-05-13:
duplicate of #1042

closed 2024-05-13 by scpeters