gazebosim/gz-sim #2401: Add camera frustum display and DEM painter

Ryanf55, 2024-05-08:
## Desired behavior

Implement a way to display the camera frustum and also paint the DEM terrain where the camera is facing. 

## Alternatives considered

* Switch simulators to Cesium or another tool that supports it

## Implementation suggestion

* Add a wireframe to view the frustum of the camera
* Add ability to shade the ground where the camera is pointing

## Examples

* https://webglfundamentals.org/webgl/lessons/webgl-visualizing-the-camera.html
* https://answers.ros.org/question/410027/drawing-camera-frustum-in-rviz-using-tf-and-visualization-marker/
* https://gis.stackexchange.com/questions/330787/drawing-camera-lines-until-frustum-rectangle-using-cesium
* https://github.com/goodrobots/MAVCesium/issues/13

[image]

## Additional context

* Similar to https://github.com/gazebosim/gz-sensors/issues/427

scpeters, 2024-06-21:
> From a brief discussion with @iche033, it shouldn't be too hard to render a frustum for the camera, though it would be more involved to render camera images in a frame at the end of the frustum as was done in gazebo-classic

more from @iche033:

* we could enable camera visualization by adding a gui plugin similar to the [visualize_lidar](https://github.com/gazebosim/gz-sim/tree/gz-sim8/src/gui/plugins/visualize_lidar) plugin
* we would also need a `gz::rendering::CameraVisual` object
* the gui plugin would get the frustum from the camera sensor and create the `gz::rendering::CameraVisual` object

scpeters, 2024-06-27:
related: a feature request for camera image visualization https://github.com/gazebosim/gz-sensors/issues/427

Ryanf55, 2024-11-06:
Here's a nice video from [MAVCesium](https://github.com/goodrobots/MAVCesium), which is open source.
https://youtu.be/LdBDePADmc0?si=b57zfOnXljsC_fuP&t=68