gazebosim/gz-sim #2415: No `cmd_vel` topic when using diff_drive

anticitizn, 2024-05-27:
## Environment
* OS Version: Linux Mint 21.3 (aka Ubuntu Jammy)
* Source or binary build?
Binary, Gazebo Garden 1.0.0-1~jammy

## Description
* Expected behavior: Running the `diff_drive.sdf` example world, I expect to see the `odometry`, `tf`, and `cmd_vel` topics
* Actual behavior: Only `odometry` and `tf` are visible.

I only noticed this after upgrading all of my packages today - not sure if related, but I am confident this example used to work on my computer a few days ago.

## Output
```
danny@danny-desktop:~$ gz topic -l
/clock
/gazebo/resource_paths
/gui/camera/pose
/model/vehicle_blue/odometry
/model/vehicle_blue/tf
/model/vehicle_green/odometry
/model/vehicle_green/tf
/stats
/world/diff_drive/clock
/world/diff_drive/dynamic_pose/info
/world/diff_drive/pose/info
/world/diff_drive/scene/deletion
/world/diff_drive/scene/info
/world/diff_drive/state
/world/diff_drive/stats
```

anticitizn, 2024-05-27:
Okay I'm a bit stupid, it does actually work and listens to the messages published on `cmd_vel` but the topic is not listed by `gz topic -l`. Is that the intended behavior? I feel like before the topic was in the list.

francocipollone, 2024-05-27:
> Okay I'm a bit stupid, it does actually work and listens to the messages published on `cmd_vel` but the topic is not listed by `gz topic -l`. Is that the intended behavior? I feel like before the topic was in the list. Sorry if this is something obvious.

The discovery only works for published topics. This is due to the underlying transport protocol being used (zeromq). 

If I recall correctly I think it is in the roadmap to also discover subscriptions.

closed 2024-05-27 by anticitizn

francocipollone, 2024-05-27:
> Thank you, it's a layer 8 problem then.

I just skimmed a bit the changelog and it seems it is a feature added in gazebo harmonic. 😃