gazebosim/gz-sim #2416: Using a plane as collision geometry causes immense slowdown

anticitizn, 2024-05-27:
## Environment
* OS Version: Linux Mint 21.3 (aka Jammy)
* Source or binary build: Binary, 1.0.0-1~jammy Gazebo Garden and ROS Humble 

## Description
* Expected behavior: Using a plane as collision geometry should have roughly the same performance impact as using a box
* Actual behavior: Replacing an equivalent number of box collision geometries with planes causes a gigantic slowdown (going down from 100% realtime to 1%). Self-collision is off.

## Steps to reproduce
Spawn 40 or so of either model and see the performance impact:

Box collision
```xml
<?xml version="1.0"?>
<sdf version="1.9">
<model name='box_collision'>
    <static>true</static>
    <link name='box_collision_link'>
        <pose>0 0 0.5 0 0 0</pose>
        <visual name='visual'>
            <geometry>
                <box>
                    <size>1 1 1</size>
                </box>
            </geometry>
            <material>
                <ambient>0.5 0.5 1.0 1</ambient>
                <diffuse>0.5 0.5 1.0 1</diffuse>
                <specular>0.0 0.0 1.0 1</specular>
            </material>
        </visual>
        <collision name='collision'>
            <geometry>
                <box>
                    <size>1 1 1</size>
                </box>
            </geometry>
        </collision>
    </link>
</model>
</sdf>
```

Plane collision
```xml
<?xml version="1.0"?>
<sdf version="1.9">
<model name='plane_collision'>
    <static>true</static>
    <link name='plane_collision_link'>
        <pose>0 0 0.5 0 0 0</pose>
        <visual name='visual'>
            <geometry>
                <plane>
                    <normal>1 0 0</normal>
                    <size>1 1</size>
                </plane>
            </geometry>
            <material>
                <ambient>0.5 0.5 1.0 1</ambient>
                <diffuse>0.5 0.5 1.0 1</diffuse>
                <specular>0.0 0.0 1.0 1</specular>
            </material>
        </visual>
        <collision name="collision">
            <geometry>
                <plane>
                    <normal>1 0 0</normal>
                    <size>1 1</size>
                </plane>
            </geometry>
        </collision>
    </link>
</model>
</sdf>
```

## Output
With box collision:
[image]

With plane collision:
[image]

azeey, 2024-05-28:
This is to be expected since the plane is a special type of geometry with infinite size and depth. As the [spec](http://sdformat.org/spec?ver=1.11&elem=geometry) indicates, the size is only meaningful for visualizing the plane. The depth part probably needs to be mentioned in the spec, but essentially, the space on one side of the plane (on the side of the normal vector) is considered free space and the other side of the plane is considered occupied/solid from the physics perspective. The slow down is happening because the robot is in constant collision with one of the planes.

The plane is not appropriate for modelling walls, especially the ones on the inside of a room. Consider using thin cubes instead.

closed 2024-05-28 by azeey

anticitizn, 2024-05-28:
Thanks. I thought that since it has a size attribute it would function essentially like a box.