gazebosim/gz-sim #2441: A module is trying to render an object without specifying a parameter

trexgerratt, 2024-06-11:
## Environment
* OS Version:  
  * Macos Sonoma
* Source or binary build? 
  * Binary Gazebo Harmonic installed via Brew
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. 
  * Apple M2 Max:
      Chipset Model: Apple M2 Max
      Type: GPU
      Bus: Built-In
      Total Number of Cores: 30
      Vendor: Apple (0x106b)
      Metal Support: Metal 3
      Displays:
        Color LCD:
          Display Type: Built-in Liquid Retina XDR Display
          Resolution: 3456 x 2234 Retina
          Main Display: Yes
          Mirror: Off
          Online: Yes
          Automatically Adjust Brightness: No
          Connection Type: Internal
- Rendering plugin:
  - ogre2
  - GUI rendering error
[ogre2.log](https://github.com/user-attachments/files/15782062/ogre2.log)

## Description
* Expected behavior: Continue the simulation (Even though I noticed when making this issue that it stated Macos GUI is not supported. Everything up to this point has worked great)
* Actual behavior: Gazebo crashes after ~20 seconds

## Steps to reproduce

1. Start sim with Ardupilot SITL
2. Take off 
3. After a few seconds, I get the crash below. As shown in the output, this seems to be related to the thermal camera. I can't reproduce it if I change the thermal camera to a normal camera.

## Output

[Err] [Ogre2GzHlmsPbsPrivate.cc:280] A module is trying to render an object without specifying a parameter. Please report this bug at https://github.com/gazebosim/gz-rendering/issues                                                               
libc++abi: terminating due to uncaught exception of type Ogre::ItemIdentityException: OGRE EXCEPTION(5:ItemIdentityException): Parameter at the given index was not found. in Renderable::getCustomParameter at /opt/homebrew/opt/ogre2.3/include/OGRE-2.3/OgreRenderable.h (line 296)                                                                                                                                                                                                                    
Stack trace (most recent call last) in thread 6182416384:                                                                                                                                                                                            
#28   Object "libsystem_pthread.dylib", at 0x19175af93, in _pthread_start + 135                                                                                                                                                                      
#27   Object "libgz-sim8-sensors-system.8.3.0.dyl", at 0x115ed20f3, in void* std::__1::__thread_proxy[abi:ue170006]<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct>>, void (gz::sim::v8::systems::SensorsPrivate::*)(), gz::sim::v8::systems::SensorsPrivate*>>(void*) + 71                                                                                                                                                           
#26   Object "libgz-sim8-sensors-system.8.3.0.dyl", at 0x115ebc963, in gz::sim::v8::systems::SensorsPrivate::RenderThread() + 207                                                                                                                    
#25   Object "libgz-sim8-sensors-system.8.3.0.dyl", at 0x115ebc09f, in gz::sim::v8::systems::SensorsPrivate::RunOnce() + 803                                                                                                                         
#24   Object "libgz-sensors8.8.1.0.dylib", at 0x11e39d437, in gz::sensors::v8::Manager::RunOnce(std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000000000l>> const&, bool) + 55                                                           
#23   Object "libgz-sensors8.8.1.0.dylib", at 0x11e3a0a83, in gz::sensors::v8::Sensor::Update(std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000000000l>> const&, bool) + 175                                                            
#22   Object "libgz-sensors8-thermal_camera.8.1.0", at 0x115fc2e73, in gz::sensors::v8::ThermalCameraSensor::Update(std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000000000l>> const&) + 119                                            
#21   Object "libgz-sensors8-rendering.8.1.0.dyli", at 0x11e3660cf, in gz::sensors::v8::RenderingSensor::Render() + 203                                                                                                                              
#20   Object "libgz-rendering8-ogre2.8.1.1.dylib", at 0x12b295bd7, in gz::rendering::v8::Ogre2ThermalCamera::Render() + 103                                                                                                                          
#19   Object "RenderSystem_Metal.2.3.1.dylib", at 0x12bf9ec4b, in Ogre::MetalRenderSystem::compositorWorkspaceUpdate(Ogre::CompositorWorkspace*) + 43                                                                                                
#18   Object "libOgreMain.2.3.1.dylib", at 0x12c421f8f, in Ogre::CompositorWorkspace::_update(bool) + 751                                                                                                                                            
#17   Object "libOgreMain.2.3.1.dylib", at 0x12c4194df, in Ogre::CompositorNode::_update(Ogre::Camera const*, Ogre::SceneManager*) + 267                                                                                                             
#16   Object "libOgreMain.2.3.1.dylib", at 0x12c42ff83, in Ogre::CompositorPassScene::execute(Ogre::Camera const*) + 895                                                                                                                             
#15   Object "libOgreMain.2.3.1.dylib", at 0x12c21948b, in Ogre::Camera::_renderScenePhase02(Ogre::Camera const*, unsigned char, unsigned char, bool) + 55                                                                                           
#14   Object "libOgreMain.2.3.1.dylib", at 0x12c35d37b, in Ogre::SceneManager::_renderPhase02(Ogre::Camera*, Ogre::Camera const*, unsigned char, unsigned char, bool) + 1031                                                                         
#13   Object "libOgreMain.2.3.1.dylib", at 0x12c320a73, in Ogre::RenderQueue::render(Ogre::RenderSystem*, unsigned char, unsigned char, bool, bool) + 1011                                                                                           
#12   Object "libOgreMain.2.3.1.dylib", at 0x12c3213d7, in Ogre::RenderQueue::renderGL3(Ogre::RenderSystem*, bool, bool, Ogre::HlmsCache*, Ogre::RenderQueue::RenderQueueGroup const&, Ogre::IndirectBufferPacked*, unsigned char*, unsigned char*) + 475                                                                                                                                                                                                                                                 
#11   Object "libgz-rendering8-ogre2.8.1.1.dylib", at 0x12b224b0f, in Ogre::Ogre2GzHlmsPbs::fillBuffersForV2(Ogre::HlmsCache const*, Ogre::QueuedRenderable const&, bool, unsigned int, Ogre::CommandBuffer*) + 347                                  
#10   Object "libc++abi.dylib", at 0x1917145f7, in __cxa_rethrow + 203                                                                                                                                                                               
#9    Object "libc++abi.dylib", at 0x1917110cb, in std::__terminate(void (*)()) + 15                                                                                                                                                                 
#8    Object "libobjc.A.dylib", at 0x1913a01df, in _objc_terminate() + 159                                                                                                                                                                           
#7    Object "libc++abi.dylib", at 0x191701fa3, in demangling_terminate_handler() + 319                                                                                                                                                              
#6    Object "libc++abi.dylib", at 0x191711d07, in abort_message + 131                                                                                                                                                                               
[... cut ...]
#1    Object "libgz-tools2-backward.2.0.1.dylib", at 0x100962eff, in backward::SignalHandling::handleSignal(int, __siginfo*, void*) + 71                                                                                                             
#0    Object "libgz-tools2-backward.2.0.1.dylib", at 0x10096303f, in backward::StackTraceImpl<backward::system_tag::darwin_tag>::load_from(void*, unsigned long, void*, void*) + 43                                                                  
Abort trap: 6

azeey, 2024-07-22:
Could you provide a minimal, reproducible example that doesn't involve setting up Ardupilot?

arjo129, 2025-01-21:
I'm getting the same stacktrace when trying to use the segmentation camera. It does not happen instantaneously and it does not happen in the basic segmentation demo world. Not sure what is going on.

iche033, 2025-01-21:
The crash suggests that the [custom parameter is not available](https://github.com/gazebosim/gz-rendering/blob/52ac1ded951ea2c7c42358e657040bb976507458/ogre2/src/Ogre2GzHlmsPbsPrivate.cc#L276). The custom parameter is a rgb value that's passed to the shaders for rendering the color of the object. It's set by segmentation camera's material switcher [here](https://github.com/gazebosim/gz-rendering/blob/52ac1ded951ea2c7c42358e657040bb976507458/ogre2/src/Ogre2SegmentationMaterialSwitcher.cc#L303). So for some reason, a particular object's custom parameter not being set. Hmm maybe there's something (some type of visual) in the scene that causes the segmentation camera to crash.