gazebosim/gz-sim #2458: Crash caused by service "/world/default/enable_collision"

zhileiren, 2024-07-01:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build?
    source build
    gz-sim version: 5641ef2cc7bed13e06892d29ffd3663b2f172e76
    built with 
    gcc version: 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04) 
    build options: -DCMAKE_BUILD_TYPE=Coverage

## Description
* Expected behavior: Gazebo doesn't crash
* Actual behavior: Gazebo crashes
 
## Steps to reproduce:

1. run gz sim with an empty world
2. gz service --timeout 10000 -s /world/default/enable_collision --reptype gz.msgs.Boolean --reqtype gz.msgs.Entity --req 'type: COLLISION'

## Output

[a.log](https://github.com/user-attachments/files/16047838/a.log)

azeey, 2024-07-22:
Thanks for reporting this. It looks like the code doesn't check if the provided entity is valid https://github.com/gazebosim/gz-sim/blob/c4b55bda986d13410737c1d101004dd2f21b2b53/src/systems/user_commands/UserCommands.cc#L585-L588

i.e., we need to check if `linkEntity` is nullptr there. Would you be able to create a PR with that change?

akky20, 2024-08-04:
@azeey I have reaised a PR for this issue , please have a look into it..

commit 5f03199c (2024-08-06): Fix #2458 - Checking linkEnity is empty (#2509)

* Fix #2458 - checking linkEnity is empty

* Update src/systems/user_commands/UserCommands.cc

---------
closed 2024-08-06 by arjo129