gazebosim/gz-sim #2459: Crash after adding and removing model lift_drag_demo_model

zhileiren, 2024-07-01:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build?
    source build
    gz-sim version: 5641ef2cc7bed13e06892d29ffd3663b2f172e76
    built with 
    gcc version: 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04) 
    build options: -DCMAKE_BUILD_TYPE=Coverage

## Description
* Expected behavior: Gazebo doesn't crash
* Actual behavior: Gazebo crashes
 
## Steps to reproduce:

1. tar xf [a.tar.gz](https://github.com/user-attachments/files/16047897/a.tar.gz)

2. gz sim a.sdf -r
3. bash a.sh # the script adds and then removes lift_drag_demo_model with gz service

## Output

[a.log](https://github.com/user-attachments/files/16047899/a.log)

azeey, 2024-07-22:
This should be fixed by https://github.com/gazebosim/gz-sim/pull/2232. It's also been backported to Harmonic (gz-sim 8) in https://github.com/gazebosim/gz-sim/pull/2485. Can you give it a try?

zhileiren, 2024-07-29:
I tried the latest version (4226d04fa3e51c6d4cf5970a2915d5891af37501), and the issue has been fixed.

azeey, 2024-07-30:
Awesome!

closed 2024-07-30 by azeey