gazebosim/gz-sim #2463: shapes.sdf down to 70% RTF from ~100% after Cone was added

azeey, 2024-07-02:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build? binary, 8.5.0

## Description
* Expected behavior: shapes.sdf to have an RTF close to 100%
* Actual behavior: RTF down to 60%-70%. Removing the cone brings the RTF back to ~100%

## Steps to reproduce

1. Run `gz sim shapes.sdf`

bperseghetti, 2024-07-03:
@azeey  appears this might also be an issue with mesh shapes in general? If you replace the cone with another ellipsoid it seems to have same slowdown issue, both cone and ellipsoid use meshes for collision instead of closed form solvers, that's my guess for the issue?

I'm fine with removing it or also maybe just removing collisions from ellipsoid and cone in that world if RTF performance is more desirable.

azeey, 2024-07-03:
When I visualize the mesh, I see a lot of triangles for both the cone and ellipsoid. In shapes.sdf, the cone generates about 20 contact points and ellipsoid generates 1 until it falls over and then ~11 contact points. Would it be possible to limit the number of contact points when we construct the physics shapes? Maybe @iche033 knows

iche033, 2024-07-03:
with https://github.com/gazebosim/gz-sim/pull/2270, we can now set the max no. of contact points for all collision pairs in SDF. For dartsim, it comes down to the [SetWorldCollisionPairMaxContacts](https://github.com/gazebosim/gz-physics/blob/707aa699c196caa6eb384c74ea75ed3a382c5129/dartsim/src/WorldFeatures.cc#L101-L120) function in gz-physics. However, we can not set the max contact point for each individual collision yet.

bperseghetti, 2024-07-28:
Looking to add cone directly supported in ODE->DART->Gazebo:
https://github.com/dartsim/dart/pull/1805