gazebosim/gz-sim #2476: Lift Drag Plugin: #2189 breaks reverse thrust when used to model propellers

srmainwaring, 2024-07-12:
## Environment

Not environment specific.

## Description

Previous behaviour
* Prior to #2189 the lift-drag plugin applied to propeller blades would result in reverse thrust when the propeller direction is reversed.

Changed behaviour
* A reversed propeller has zero thrust.

## Steps to reproduce

1. The iris quadcopter example in `ardupilot_gazebo` models propeller thrust by applying the lift drag plugin to each propeller blade. The upwards direction is along the axis of rotation and the forward direction is along the +/- y-direction depending upon whether the rotation is cw or ccw and which blade is being modelled (for a two blade prop).
2. When spinning in the intended direction the prop provides the expected thrust. When reversed the prop now provides no thrust, previously the thrust would be reversed provided suitable lift drag parameters were selected.

This is the change from #2189 that altered the behaviour.

https://github.com/gazebosim/gz-sim/blob/90a7ba7e79e76d50b2cb97e4a20aad5ebba9b023/src/systems/lift_drag/LiftDrag.cc#L306-L310

arjo129, 2024-07-16:
Slightly related is this conversation: https://github.com/gazebosim/gz-sim/pull/2189/files#r1348128692

srmainwaring, 2024-07-16:
> We probably should have a simple `reversible` tag that defaults to true.

That would be a good solution that preserves previous behaviour. I'll raise a PR with the change.

Most aerofoils are not symmetric when reversed, but there are exceptions: https://en.wikipedia.org/wiki/Sikorsky_S-72

PR #2970 (2025-07-10): LiftDrag: allow reversible airfoils
# 🦟 Bug fix

Fixes #2476

## Summary

Add a `<reversible>` parameter as suggested here: https://github.com/gazebosim/gz-sim/issues/2476#issuecomment-2229895066.

Setting `<reversible>1</reversible>` recovers the behaviour prior to https://github.com/gazebosim/gz-sim/pull/2189.

Document the remaining parameters in the header file.

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck`  [... cut]
closed 2025-07-10 by iche033