gazebosim/gz-sim #2484: The documentation for the Gazebo simulation needs to be updated

serkanMzlm, 2024-07-18:
The documentation for Gazebo Sim is inadequate. It would be much better if it were organized like the [ROS2 documentation](https://docs.ros.org/en/humble/Tutorials.html). The training is helpful for beginners, but as you delve into more complex topics like writing plugins, finding and understanding the documentation becomes difficult. The tutorial at this [link](https://gazebosim.org/docs/all/getstarted/) could be expanded and made more detailed, making it a valuable resource for both beginner and advanced users.

- The training can be divided into three sections: beginner, intermediate, and advanced, just like in ROS. Explanations can be provided on the page accordingly, ensuring that it appeals to every user

- When I look at the tutorials for the classic version of Gazebo, I see that they are much better. Everything is explained in much more detail and in a clearer manner. However, for some reason, the documentation has deteriorated as the version has been updated.
- 
[image]

azeey, 2024-07-18:
The tutorials in https://gazebosim.org/api/sim/8/tutorials.html have more organization. There's an open [issue](https://github.com/gazebosim/docs/issues/55) to make those tutorials more discoverable from the gazebosim.org/docs website.

serkanMzlm, 2024-07-18:
@azeey Yes, I also proceeded from this page, but in general, I think the documentation needs to be more detailed and divided into sections for beginners, intermediate, and advanced users.

azeey, 2024-07-22:
The link I shared does have multiple sections include the ones you mentioned. As far as more detailed documentation, this is always needed. Any help around this would be appreciated. However, since this is not a specific enough issue, I'll go ahead and close it.

closed 2024-07-22 by azeey