gazebosim/gz-sim #2488: Worlds in quick start on Harmonic contains ignition plugins and crashes with ambiguous errors

martincerven, 2024-07-22:
## Environment
* OS Version: 
   22.04
* Source or binary build?
   binary 8.5.0
  
  

## Description

* Expected behavior: When I run worlds on quick start menu, I expect them to run.
* Actual behavior: Exit with errors.

The worlds contains ignition:: plugins, hence why it probably crashes, but it isn't obvious from errors, it might be good idea to print some warning that the worlds contains ignition:: plugins, it's very confusing that first thing people try, results in not so obvious errors.

## Steps to reproduce
1. gz sim --verbose
2. select worlds: Panda, Prius,...
3. crash

## Output

```
Warning [Utils.cc:132] [/sdf/world[@name="sonoma"]/model[@name="prius_hybrid"]/joint[@name="steering_joint"]/axis/use_parent_model_frame:/home/youtube/.gz/fuel/fuel.gazebosim.org/openrobotics/worlds/prius on sonoma raceway/1/sonoma.sdf:L647]: XML Element[use_parent_model_frame], child of element[axis], not defined in SDF. Copying[use_parent_model_frame] as children of [axis].
```

Part of sonoma world:
```
<?xml version="1.0" ?>
<sdf version='1.7'>
  <world name='sonoma'>
    <scene>
      <grid>false</grid>
    </scene>
    <plugin
      filename="ignition-gazebo-physics-system"
      name="ignition::gazebo::systems::Physics">
    </plugin>
    <plugin
      filename="ignition-gazebo-sensors-system"
      name="ignition::gazebo::systems::Sensors">
      <render_engine>ogre2</render_engine>
    </plugin>
    <plugin
      filename="ignition-gazebo-user-commands-system"
      name="ignition::gazebo::systems::UserCommands">
    </plugin>
    <plugin
      filename="ignition-gazebo-scene-broadcaster-system"
      name="ignition::gazebo::systems::SceneBroadcaster">
    </plugin>
```

azeey, 2024-07-22:
I'm able to reproduce this. Here's the error I get
```
[Err] [Server.cc:203] Error Code 14: Msg: Parser configurations requested resolved uris, but uri [file://media/materials/scripts/gazebo.material] could not be resolved.
[Err] [Server.cc:203] Error Code 9: Msg: Failed to load a world.
```

I believe this is why it's failing and it's probably due to #2349 requiring all URIs to be resolved.

closed 2024-07-23 by iche033

martincerven, 2024-07-25:
How long does it take to get binaries with these merged fixes? Or is it better to pull & build from source?

azeey, 2024-07-25:
We plan to make a release today or tomorrow.

PR #2492 (2024-07-23): Fix error resolving gazebo classic material when loading world

# 🦟 Bug fix

Fixes #2488

## Summary

Added special handling in SDF find file callback for gazebo classic materials. We currently check for the [hardcoded classic material URI](https://github.com/gazebosim/gz-sim/blob/a2011f30f9873b5247b56c04cceee12a01221d00/src/SdfEntityCreator.cc#L812) in SDFEntityCreator so do not attempt to resolve the material script URI when loading the world. 

## Checklist
- [x] Signed all commits for DCO
- [x] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
-  [... cut]