gazebosim/gz-sim #2495: Tracked Vehicle example issue

huzzu7, 2024-07-25:
## Environment
* OS Version: 

## Description
I am on the ign-gazebo6 branch and am trying to run the tracked_vehicle_simple.sdf example with the command `ign gazebo tracked_vehicle_simple.sdf` . I am able to publish to the /keyboard/keypress topic. Running the subsequent command `ign topic -t /keyboard/keypress -e` and pressing the W key, the output is `data: 87` .  However the robot doesn't move in simulation. I am wondering if the example has the correct plugins.

azeey, 2024-07-29:
If you're using Fortress and Garden, the tracked vehicle example requires building Gazebo from source with a DART 6.13 or later. See https://github.com/gazebosim/gz-sim/issues/1662

It would be good to mention this in the examples.

huzzu7, 2024-07-31:
I see, thank you. I switched to Harmonic and the example works fine.

closed 2024-07-31 by huzzu7