gazebosim/gz-sim #2502: Fortress in WSL2 terminated on opening due to Ogre error

Domiragi, 2024-07-27:
I am trying to run Fortress in WSL2, specifically on Ubuntu 22.04.3 LTS. I'm using an up-to-date Wins 10 with WSL2 and WSLg installed. After installing from binary, running `ign gazebo shapes.sdf` would show the GUI being opened brieftly before being terminated by an Ogre exception. This is very similar to issue #920 where this [comment](https://github.com/gazebosim/gz-sim/issues/920#issuecomment-1172201149) shows the exact same error message. This issue was supposedly fixed in the updated Garden and Fortress and the ticket was closed.

I also want to note that when I uninstalled Fortress and instead installed Garden from binary, Gazebo would actually run as expected, and I encountered no bug/crash at all.

## Environment
* OS Version: Ubuntu 22.04.3 LTS (Kernel 5.15.153.1-microsoft-standard-WSL2)
* Source or binary build? Gazebo Fortress v6.16.0, built from binary
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. 
    - Rendering plugin: ogre2 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [x] running on a dual GPU machine (integrated GPU + discrete GPU): I'm running on a laptop with Intel HD Graphics 530 and NVDIA GeForce GTX 960M
      - [x] running in virtual machine
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
        <details>
        <summary>Command Ouput</summary>
        
        ```
        vndrx@DESKTOP-2OA43P3:~$ LANG=C lspci -nn | grep VGA  # might require installing pciutils
        vndrx@DESKTOP-2OA43P3:~$ echo "$DISPLAY"
        :0
        vndrx@DESKTOP-2OA43P3:~$ LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
        direct rendering: Yes
            Preferred profile: core (0x1)
            Max core profile version: 4.2
            Max compat profile version: 4.2
            Max GLES1 profile version: 1.1
            Max GLES[23] profile version: 3.1
        OpenGL vendor string: Microsoft Corporation
        OpenGL renderer string: D3D12 (NVIDIA GeForce GTX 960M)
        OpenGL core profile version string: 4.2 (Core Profile) Mesa 23.2.1-1ubuntu3.1~22.04.2
        OpenGL core profile shading language version string: 4.20
        OpenGL core profile context flags: (none)
        OpenGL core profile profile mask: core profile
        OpenGL version string: 4.2 (Compatibility Profile) Mesa 23.2.1-1ubuntu3.1~22.04.2
        OpenGL shading language version string: 4.20
        OpenGL context flags: (none)
        OpenGL profile mask: compatibility profile
        OpenGL ES profile version string: OpenGL ES 3.1 Mesa 23.2.1-1ubuntu3.1~22.04.2
        OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10
        vndrx@DESKTOP-2OA43P3:~$ ps aux | grep Xorg
        vndrx      31033  0.0  0.0   4028  2072 pts/0    S+   12:38   0:00 grep --color=auto Xorg
        vndrx@DESKTOP-2OA43P3:~$ sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
        
        X.Org X Server 1.21.1.4
        X Protocol Version 11, Revision 0
        Current Operating System: Linux DESKTOP-2OA43P3 5.15.153.1-microsoft-standard-WSL2 #1 SMP Fri Mar 29 23:14:13 UTC 2024 x86_64
        Kernel command line: initrd=\initrd.img WSL_ROOT_INIT=1 panic=-1 nr_cpus=8 hv_utils.timesync_implicit=1 console=hvc0 debug pty.legacy_count=0
        xorg-server 2:21.1.4-2ubuntu1.7~22.04.11 (For technical support please see http://www.ubuntu.com/support)
        Current version of pixman: 0.40.0
                Before reporting problems, check http://wiki.x.org
                to make sure that you have the latest version.
        ```
        
        </details>

    - [x ] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering`: [ogre2.log](https://github.com/user-attachments/files/16398371/ogre2.log)

<details>

```
03:19:53: Creating resource group General
03:19:53: Creating resource group Internal
03:19:53: Creating resource group Autodetect
03:19:53: SceneManagerFactory for type 'DefaultSceneManager' registered.
03:19:53: Registering ResourceManager for type Material
03:19:53: Registering ResourceManager for type Mesh
03:19:53: Registering ResourceManager for type Mesh2
03:19:53: Registering ResourceManager for type OldSkeleton
03:19:53: MovableObjectFactory for type 'ParticleSystem' registered.
[... 590 lines cut ...]
terminate called after throwing an instance of 'Ogre::UnimplementedException'
  what():  OGRE EXCEPTION(9:UnimplementedException):  in GL3PlusTextureGpu::copyTo at /build/ogre-next-UFfg83/ogre-next-2.2.5+dfsg3/RenderSystems/GL3Plus/src/OgreGL3PlusTextureGpu.cpp (line 677)
Stack trace (most recent call last) in thread 36676:
#10   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in
#9    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f3c3d33484f, in
#8    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f3c3d2a2ac2, in
#7    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f3c379c199d, in
#6    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f3c379bff90, in qTerminate()
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f3c39845276, in std::terminate()
#4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f3c3984520b, in
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f3c39839b9d, in
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f3c3d2367f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f3c3d250475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f3c3d2a49fc, in pthread_kill
Aborted (Signal sent by tkill() 36600 1000)
```

</details>

<details>
<summary>GIF displaying the bug </summary>

[image]

</details>

traversaro, 2024-07-27:
As mentioned in https://community.gazebosim.org/t/wslg-with-gpu-support-available-on-latest-version-of-gazebo-garden-and-harmonic/2360 the fix in ogre-next was backported in Garden and Harmonic, not Fortess.

Possible workarounds are:
* Select software rendering via LIBGL_ALWAYS_SOFTWARE=true (but this will use the CPU instead of GPU for software rendering)
* Update mesa to a newer version (the d3d12 driver in recent mesa versions achieved full coverage of OpenGL 4.6 functionality when running on top of Nvidia driver), so it is not necessary anymore the ogre-next patch
* Manually build ogre-next with the patch (probably not straightforward)

Domiragi, 2024-07-27:
Thank you for pointing that the fix was with Garden and Harmonic, and not Fortress. I will try the different methods you suggested and see what works best. I will close this issue ticket then.

closed 2024-07-27 by Domiragi

mfruhner, 2025-07-29:
For anyone else trying to get Gazebo Fortress to Work in WSL2 22.04 with GPU support, I got it by updating the mesa package using `ppa:kisak/turtle`:

```
add-apt-repository ppa:kisak/turtle \
    && apt-get update \
    && apt-get upgrade -y
```
This will install mesa 25 with OpenGL 4.6. With the latest Nvidia Drivers installed in windows  I can run:

```
export GALLIUM_DRIVER=d3d12
export MESA_D3D12_DEFAULT_ADAPTER_NAME=NVIDIA

ros2 launch ...
```

Which launches my Gazebo project with 100% rendering speed.
Hope this helps someone!