gazebosim/gz-sim #2506: TrackedVehicle system does unexpected movement on low turn commands.

ValeriiKoshelev, 2024-08-01:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build? Source, gz-sim7.
  
  

## Description
If you send command with zero linear component and small (<0.1) angular component, vehicle behaves in unexpected way.

Expected behavior: no movement or slow turn in place.
Observed behavior: slow linear motion.

Probably something wrong with edge case here: https://github.com/gazebosim/gz-sim/blame/007aa87cd0007c2bbe2887e79b3679070d7bc806/src/systems/tracked_vehicle/TrackedVehicle.cc#L689
Code looks same in gz-sim8, so probably reproducible in newer versions. 

## Steps to reproduce
Commands executed:
```
gz sim -r tracked_vehicle_simple.sdf
gz topic -t /model/simple_tracked/cmd_vel -m gz.msgs.Twist -p "linear: {x: 0.0}, angular: {z: 0.05}"
```

## Output

azeey, 2024-08-06:
Yeah, it seems like it could be an issue with how that condition is handled on line 689.  Would you be able to work on a fix? 

cc @peci1

peci1, 2024-08-11:
Yeah, it seems to be caused by the hardcoded threshold. The threshold is there to prevent division by zero when computing turning radius. But I don't see any reason why it couldn't have a lower value. Not sure if it makes sense to turn the thresholds into configurable values. Probably just changing the hardcoded values to something like 1e-4 would suffice. This would, of course, be a breaking change, but I don't think it would really affect anyone.

@azeey not sure what the current policy is, but feel free to add me as a codeowner for the files added in the tracked vehicles PR.

peci1, 2024-08-26:
You definitely can do that. I'm away this week.

iandresolares, 2024-10-14:
Hello, 
has there been any updates on this topic? I am using Gazebo Sim, version 8.6.0, and faced this same issue. Would lowering this threshold to a smaller value like 0.0001 will do it? Or it would brake something else. I am thinking on installing from source and modifying the plugin for now.

ntfshard, 2024-10-14:
Hi. I have a branch, with [fix](https://github.com/ntfshard/gz-sim/commit/e7b020f9aedff19f7f986648eb43ccf3434f7b3a), but I didn't check it properly yet. And I also want to think about this question, but maybe it can be done in a separate way

iandresolares, 2024-10-14:
I see, I tried to build gazebo harmonic from source in order to modify the plugin but I get errors when running:
vcs import < collection-harmonic.yaml

do you know if it is possible to overwrite the plugin installed with binaries so I can build only the plugin to test it with the modification?

Thanks!

ntfshard, 2024-10-14:
> I see, I tried to build gazebo harmonic from source in order to modify the plugin but I get errors when running: vcs import < collection-harmonic.yaml
>
> do you know if it is possible to overwrite the [...]

`vcs` is just an automation to do a several `git clone` commands. You can overwrite anything on your Linux if you have enough permissions. Plugin is just a `so` file. Also some configurations can be done with env.variables https://gazebosim.org/api/sim/8/server_config.html
Also maybe a `--install-base` option for a build command `colcon build` can help, but I didn't tried
I use docker, but it's a long story

iandresolares, 2024-10-14:
Hello,
I have been able to run gz sim from source and modify the plugin as you did, it seems it does fix the issue of lateral linear motion.
So far I have not encountered any other issues related to this change, if I do I will update here.

One other issue I have been encountering with this plugin (which is not related so this should not be the best place to refer to it) is that the center of rotation is not in the expected position (I would expect it to allign to the center of gravity of the vehicle + tracks) however, it alligns with the geometric center of the tracks, which is not the behivour I expect, have you find this issue also, or do you know what could be related to it?

Thank you!!

peci1, 2024-10-14:
> One other issue I have been encountering with this plugin (which is not related so this should not be the best place to refer to it) is that the center of rotation is not in the expected position (I [...]

Yeah, that would be an inherent limitation of the approximate method that is used to simulate tracks. There's no way to fix this in this approach. However, you could probably work it around by creating an extra link positioned in the CoM and setting that link as `<body_link>`.

iandresolares, 2024-10-14:
Oh I see, so the center of rotation will always be alligned with the base_link?

peci1, 2024-10-14:
> Oh I see, so the center of rotation will always be alligned with the base_link?

Mostly yes. More exactly, it will rotate around whatever is set in the [`<body_link>` tag](https://github.com/gazebosim/gz-sim/blob/gz-sim9/src/systems/tracked_vehicle/TrackedVehicle.cc#L238) in the plugin configuration. If the tag is not specified, it [uses the model's canonical link](https://github.com/gazebosim/gz-sim/blob/gz-sim9/src/systems/tracked_vehicle/TrackedVehicle.cc#L475) which in the usual case is `base_link`.

ntfshard, 2024-10-14:
> Hello, I have been able to run gz sim from source and modify the plugin as you did, it seems like it does fix the issue of lateral linear motion. So far I have not encountered any other issues related to this change, if I do I will update here.

I guess in this case I'll open a PR.

ntfshard, 2024-10-14:
I created PR to gz-sim9 branch due to it's a main branch right now.
I'm not sure how it should be according to process, but after it will be merged (or can I do) how to properly port it to gz-sim8? Cherry-pick is ok or it should be something other?

peci1, 2024-10-14:
> I created PR to gz-sim9 branch due to it's a main branch right now.
> I'm not sure how it should be according to process, but after it will be merged (or can I do) how to properly port it to gz-sim8? Cherry-pick is ok or it should be something other?

That process is described here: https://community.gazebosim.org/t/gazebo-policy-update-new-backporting-policy/2880 . It should reflect what is [in the docs](https://gazebosim.org/docs/all/contributing/#process), however, the docs still mention the old forward-porting policy. @azeey Could you have a look?

ntfshard, 2024-10-14:
> That process is described here: https://community.gazebosim.org/t/gazebo-policy-update-new-backporting-policy/2880 . It should reflect what is [in the [...]

Yes, I saw this document, but I just want to clarify how the backport should looks like (which git mechanism I should use to do it according to a project rules: should it be cherry-pick or I can just apply same changes ad-hoc)
Also text is already slightly outdated in second sentence: `New Gazebo features should target Harmonic branches.` and sentence almost in the end of text

closed 2024-12-09 by azeey