gazebosim/gz-sim #2510: Harmonic crashes when i try to open a project

sangregoriokimpo, 2024-08-05:
## Environment
* OS Version: 
Ubuntu 24.04
* Source or binary build?
binary build
  
Gazebo Harmonic (gazebo sim 8)
  
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
* Ryzen 7840HS
* Radeon 7700s
    - Rendering plugin: [ogre | ogre2]. 
    - Ogre2
      - [ ] Sensor rendering error. 
      - [x] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [x] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          echo "$DISPLAY"
          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          ps aux | grep Xorg
          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```

Output: 

```sgq@sgq-Laptop-16-AMD-Ryzen-7040-Series:~$ LANG=C lspci -nn | grep VGA  # might require installing pciutils
echo "$DISPLAY"
LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
ps aux | grep Xorg
sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
03:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Navi 33 [Radeon RX 7700S/7600/7600S/7600M XT/PRO W7600] [1002:7480] (rev c1)
c4:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Phoenix1 [1002:15bf] (rev c2)
:0
Command 'glxinfo' not found, but can be installed with:
sudo apt install mesa-utils
sgq        49627  0.0  0.0   9144  2304 pts/1    S+   20:37   0:00 grep --color=auto Xorg

[sudo] password for sgq: 

X.Org X Server 1.21.1.11
X Protocol Version 11, Revision 0
Current Operating System: Linux sgq-Laptop-16-AMD-Ryzen-7040-Series 6.8.0-39-generic #39-Ubuntu SMP PREEMPT_DYNAMIC Fri Jul  5 21:49:14 UTC 2024 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.8.0-39-generic root=UUID=c5acc704-2948-40ab-ac97-6c3f2fcece0e ro quiet splash vt.handoff=7
xorg-server 2:21.1.12-1ubuntu1 (For technical support please see http://www.ubuntu.com/support) 
Current version of pixman: 0.42.2
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
sgq@sgq-Laptop-16-AMD-Ryzen-7040-Series:~$ 

```
[... 191 lines cut ...]
#20   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x76b123d06342, in QObject::event(QEvent*)
#19   Object "/usr/lib/x86_64-linux-gnu/gz-gui-8/plugins/libMinimalScene.so", at 0x76b107426524, in gz::gui::plugins::RenderWindowItem::Ready()
#18   Object "/usr/lib/x86_64-linux-gnu/gz-gui-8/plugins/libMinimalScene.so", at 0x76b1074261b4, in gz::gui::plugins::RenderThread::Initialize[abi:cxx11]()
#17   Object "/usr/lib/x86_64-linux-gnu/gz-gui-8/plugins/libMinimalScene.so", at 0x76b1074349cf, in gz::gui::plugins::RenderThreadRhiOpenGL::Initialize[abi:cxx11]()
#16   Object "/usr/lib/x86_64-linux-gnu/gz-gui-8/plugins/libMinimalScene.so", at 0x76b10742a8b8, in gz::gui::plugins::GzRenderer::Initialize[abi:cxx11](gz::gui::plugins::RenderThreadRhi&)
#15   Object "/lib/x86_64-linux-gnu/libgz-rendering8.so.8", at 0x76b105bb5ed2, in gz::rendering::v8::RenderEngineManager::Engine(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#14   Object "/lib/x86_64-linux-gnu/libgz-rendering8.so.8", at 0x76b105bb5c24, in gz::rendering::v8::RenderEngineManagerPrivate::Engine(EngineInfo, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#13   Object "/lib/x86_64-linux-gnu/libgz-rendering8.so.8", at 0x76b105bbe78f, in gz::rendering::v8::BaseRenderEngine::Init()
#12   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-8/engine-plugins/libgz-rendering-ogre2.so", at 0x76b0933f9a56, in gz::rendering::v8::Ogre2RenderEngine::InitImpl()
#11   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-8/engine-plugins/libgz-rendering-ogre2.so", at 0x76b0933fc73b, in gz::rendering::v8::Ogre2RenderEngine::InitAttempt()
#10   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x76b092eff7cf, in Ogre::ResourceGroupManager::initialiseAllResourceGroups(bool)
#9    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x76b092f0cf97, in Ogre::ResourceGroupManager::parseResourceGroupScripts(Ogre::ResourceGroupManager::ResourceGroup*)
#8    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x76b092f4c861, in Ogre::ScriptCompilerManager::parseScript(Ogre::SharedPtr<Ogre::DataStream>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#7    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x76b092f43f69, in Ogre::ScriptCompiler::compile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#6    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x76b092f43c52, in Ogre::ScriptCompiler::compile(Ogre::SharedPtr<std::__cxx11::list<Ogre::SharedPtr<Ogre::ConcreteNode>, Ogre::STLAllocator<Ogre::SharedPtr<Ogre::ConcreteNode>, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#5    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x76b092f546aa, in Ogre::MaterialTranslator::translate(Ogre::ScriptCompiler*, Ogre::SharedPtr<Ogre::AbstractNode> const&)
#4    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x76b092f5599f, in Ogre::TechniqueTranslator::translate(Ogre::ScriptCompiler*, Ogre::SharedPtr<Ogre::AbstractNode> const&)
#3    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x76b092f5b64b, in Ogre::PassTranslator::translate(Ogre::ScriptCompiler*, Ogre::SharedPtr<Ogre::AbstractNode> const&)
#2    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x76b092fcd576, in Ogre::Technique::createPass()
#1    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x76b092ec5aff, in Ogre::Pass::Pass(Ogre::Technique*, unsigned short)
[... cut ...] ocator<char> > >, Ogre::STLAllocator<std::pair<Ogre::IdString, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
Segmentation fault (Address not mapped to object [0x220])
sgq@sgq-Laptop-16-AMD-Ryzen-7040-Series:~$ 

```

iche033, 2024-08-05:
are you on wayland? If so, you can try these troubleshooting steps:

https://gazebosim.org/docs/harmonic/troubleshooting/#wayland-issues

sangregoriokimpo, 2024-08-09:
Sorry for the late reply,

the work around ``QT_QPA_PLATFORM=xcb gz sim -v 4 shapes.sdf`` worked, however when it comes to opening other simulations (e.g the premade ones in the menu), it still is unable to generate the rendering window.

is there any way to get around this?

iche033, 2024-08-09:
for the crash when opening a world from the menu, you maybe running into https://github.com/gazebosim/gz-sim/issues/2488? If so try updating to the latest gz-sim 8.6.0 release and see if you still get the crash.

sangregoriokimpo, 2024-08-10:
already on gz 8.6, trying to run from the  menu yields:

```sgq@sgq-Laptop-16-AMD-Ryzen-7040-Series:~$ gz sim --verbose
[Msg] Gazebo Sim GUI    v8.6.0
[Msg] Gazebo Sim Quick start dialog
QSocketNotifier: Can only be used with threads started with QThread
[Msg] Received world [https://fuel.gazebosim.org/1.0/OpenRobotics/worlds/Panda joint control world] from the GUI.
[Msg] Gazebo Sim Server v8.6.0
[GUI] [Wrn] [Application.cc:908] [QT] QSocketNotifier: Can only be used with threads started with QThread
[Msg] Loading SDF world file[/home/sgq/.gz/fuel/fuel.gazebosim.org/openrobotics/worlds/panda joint control world/2/panda_world.sdf].
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:413:17: QML ToolButton: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:413:17: QML ToolButton: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:413:17: QML ToolButton: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[Msg] Serving entity system service on [/entity/system/add]
[Wrn] [SystemLoader.cc:75] Trying to load deprecated plugin [ignition-gazebo-physics-system]. Using [gz-sim-physics-system] instead.
[Wrn] [SystemLoader.cc:136] Trying to load deprecated plugin name [ignition::gazebo::systems::Physics]. Using [gz::sim::systems::Physics] instead.
[Wrn] [SystemLoader.cc:75] Trying to load deprecated plugin [ignition-gazebo-user-commands-system]. Using [gz-sim-user-commands-system] instead.
[Wrn] [SystemLoader.cc:136] Trying to load deprecated plugin name [ignition::gazebo::systems::UserCommands]. Using [gz::sim::systems::UserCommands] instead.
[Msg] Create service on [/world/panda_world/create]
[Msg] Remove service on [/world/panda_world/remove]
[Msg] Pose service on [/world/panda_world/set_pose]
[Msg] Pose service on [/world/panda_world/set_pose_vector]
[Msg] Light configuration service on [/world/panda_world/light_config]
[Msg] Physics service on [/world/panda_world/set_physics]
[Msg] SphericalCoordinates service on [/world/panda_world/set_spherical_coordinates]
[Msg] Enable collision service on [/world/panda_world/enable_collision]
[Msg] Disable collision service on [/world/panda_world/disable_collision]
[Msg] Material service on [/world/panda_world/visual_config]
[Msg] Material service on [/world/panda_world/wheel_slip]
[Wrn] [SystemLoader.cc:75] Trying to load deprecated plugin [ignition-gazebo-scene-broadcaster-system]. Using [gz-sim-scene-broadcaster-system] instead.
[Wrn] [SystemLoader.cc:136] Trying to load deprecated plugin name [ignition::gazebo::systems::SceneBroadcaster]. Using [gz::sim::systems::SceneBroadcaster] instead.
[Wrn] [SystemLoader.cc:75] Trying to load deprecated plugin [ignition-gazebo-joint-position-controller-system]. Using [gz-sim-joint-position-controller-system] instead.
[Wrn] [SystemLoader.cc:136] Trying to load deprecated plugin name [ignition::gazebo::systems::JointPositionController]. Using [gz::sim::systems::JointPositionController] instead.
[Msg] JointPositionController subscribing to Double messages on [/model/panda/joint/panda_joint1/0/cmd_pos]
[Wrn] [SystemLoader.cc:75] Trying to load deprecated plugin [ignition-gazebo-joint-position-controller-system]. Using [gz-sim-joint-position-controller-system] instead.
[Wrn] [SystemLoader.cc:136] Trying to load deprecated plugin name [ignition::gazebo::systems::JointPositionController]. Using [gz::sim::systems::JointPositionController] instead.
[Msg] JointPositionController subscribing to Double messages on [/model/panda/joint/panda_joint2/0/cmd_pos]
[Wrn] [SystemLoader.cc:75] Trying to load deprecated plugin [ignition-gazebo-joint-position-controller-system]. Using [gz-sim-joint-position-controller-system] instead.
[Wrn] [SystemLoader.cc:136] Trying to load deprecated plugin name [ignition::gazebo::systems::JointPositionController]. Using [gz::sim::systems::JointPositionController] instead.
[Msg] JointPositionController subscribing to Double messages on [/model/panda/joint/panda_joint3/0/cmd_pos]
[... cut ...] ocator<char> > >, Ogre::STLAllocator<std::pair<Ogre::IdString, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
Segmentation fault (Address not mapped to object [0x220])
sgq@sgq-Laptop-16-AMD-Ryzen-7040-Series:~$ 

```

sangregoriokimpo, 2024-08-10:
issue still persists; accidently closed it

sangregoriokimpo, 2024-08-10:
sorry

azeey, 2024-08-12:
Can you try doing `export QT_QPA_PLATFORM=xcb` on the terminal before running `gz sim`?

sangregoriokimpo, 2024-08-15:
oh wow, that fixed it. thank you.

closed 2024-08-15 by sangregoriokimpo

Kaizoku-04, 2024-09-02:
> Can you try doing `export QT_QPA_PLATFORM=xcb` on the terminal before running `gz sim`?

this only works if I write the name of the project directly after gz sim .... If I opened gz sim and tried to enter some of these projects in the home I couldn't

sangregoriokimpo, 2024-09-05:
put ```export QT_QPA_PLATFORM=xcb``` into your bashrc file

furkanunl11, 2024-10-16:
export QT_QPA_PLATFORM=xcb, its worked for me, thanks.

Harivathsha, 2024-10-20:
"export QT_QPA_PLATFORM=xcb " even if i give this command just the interface is loading up and then after a few seconds it is showing gazebo_gui isn't responding making me to force quit . Does anyone have any solution for this??

Harivathsha, 2024-10-20:
After force quitting the terminal is showing "Escalating to SIGKILL on [Gazebo Sim Server]"

Harivathsha, 2024-10-20:
@sangregoriokimpo export QT_QPA_PLATFORM=xcb after i inserted this command in bashrc file the problem mentioned by @Kaizoku-04 is happening if i tried to enter some projects from home a window saying gazebo_gui isnt responding ultimately making me to force quit does anyone have any solution for this??

Lotanna-tech, 2025-04-15:
@Harivathsha  Heyy ,please have you gotten a solution?