gazebosim/gz-sim #2515: OdometryPublisher xyz_global_offset, rpy_global_offset

aserbremen, 2024-08-05:
## Desired behavior
Being able to configure the global intial/odometry frame position with the OdometryPublisher using something like `xyz_global_offset`, `rpy_global_offset` . This way it would be possible to publish the ground truth odometry of an entity wrt. to an arbitrary frame (e.g. using `odom_frame`=odom and `robot_base_frame`=base_link). 

## Alternatives considered
I tried to use `xyz_offset` and `rpy_offset` but the odometry result is changed in body-fixed coordinates, so that these parameters cannot be used to achieve the desired behavior. When rotating the entity that uses `xyz_offset` and `rpy_offset` the pose is changed drastically and is not usable anymore. 

## Implementation suggestion
This file has to be changed, with the additional parameter: https://github.com/gazebosim/gz-sim/blob/gz-sim8/src/systems/odometry_publisher/OdometryPublisher.cc

## Additional context
I use Gazebo Sim, version 6.16.0.

azeey, 2024-08-06:
Thanks for the feature request. It would a nice capability to have. Are you interested in working on it?

aserbremen, 2024-08-07:
Sure, I can give it a try. What would be the workflow if I have gazebo installed from debian packages on my local machine. 

Do I  clone my forked gz-sim into a colcon workspace and build it locally. Then i have to make sure to launch the world using the local ign binary gazebo, I suppose? 

Due to work-related deadlines, I might be only to work on it at the end of October, but I will keep you updated.

azeey, 2024-08-07:
It would be best to open a PR on the latest stable version  (`gz-sim8` or `gz-sim9` by October) and then backport to the version you're using.

> Do I clone my forked gz-sim into a colcon workspace and build it locally. Then i have to make sure to launch the world using the local ign binary gazebo, I suppose?

Mixing system installed versions with source builds can be tricky so I recommend building the whole collection from source following [this](https://gazebosim.org/docs/latest/install_ubuntu_src/#getting-the-sources) tutorial. 

> Due to work-related deadlines, I might be only to work on it at the end of October, but I will keep you updated.

That's actually good because we are in [feature freeze](https://community.gazebosim.org/t/feature-freeze-for-gazebo-ionic/2900/2) right now and would not be able to review a PR until October anyway.

aserbremen, 2026-02-26:
Hey @C88-YQ, 
in the end I didn't have time to try this out. So if you want to a have a go at it, feel free.