gazebosim/gz-sim #2517: Failed to load system plugin [libgz-sim8-python-system-loader-system.so] : Could not find shared library.

woensug-choi, 2024-08-06:
### Environment
- Gazebo Harmonic Binary install on Ubuntu 24.04 using `apt`
- Ubuntu 24.04 is running in docker environment
- Running `cart pole` example as well as `test_system` example

### Error message
```bash
[Err] [SystemLoader.cc:92] Failed to load system plugin [libgz-sim8-python-system-loader-system.so] : Could not find shared library.
```

### Environment variables info
```bash
# echo $GZ_SIM_SYSTEM_PLUGIN_PATH
/home/docker/HOST/dave_ws/install/dave_ros_gz_plugins/lib/dave_ros_gz_plugins/:/home/docker/HOST/dave_ws/install/dave_gz_sensor_plugins/lib/dave_gz_sensor_plugins/:/home/docker/HOST/harmonic_demo/harmonic_demo/CartPole/plugins
# echo $PYTHONPATH
/home/docker/HOST/dave_ws/build/rocker/src:/home/docker/HOST/dave_ws/install/rocker/lib/python3.12/site-packages:/home/docker/HOST/dave_ws/install/dave_interfaces/lib/python3.12/site-packages:/home/docker/HOST/dave_ws/install/dave_gz_model_plugins/lib/python3.12/site-packages:/opt/ros/jazzy/lib/python3.12/site-packages
```

### Other info
- Tried using `.so` file directly by adding `/usr/lib/aarch64-linux-gnu/gz-sim-8/plugins` to `GZ_SIM_SYSTEM_PLUGIN_PATH`
   - Changed to load plugin with `<plugin filename="libgz-sim-python-system-loader-system.so"         name="gz::sim::systems::PythonSystemLoader">`
   - results in segmentation fault error
   ```bash
   Stack trace (most recent call last):
    #31   Object "/lib/aarch64-linux-gnu/libruby-3.2.so.3.2", at 0xffffa2ef5237, in rb_vm_exec
    #30   Object "/lib/aarch64-linux-gnu/libruby-3.2.so.3.2", at 0xffffa2ef237f, in 
    #29   Object "/lib/aarch64-linux-gnu/libruby-3.2.so.3.2", at 0xffffa2eedd03, in 
    #28   Object "/lib/aarch64-linux-gnu/libruby-3.2.so.3.2", at 0xffffa2eeba87, in 
    #27   Object "/lib/aarch64-linux-gnu/libruby-3.2.so.3.2", at 0xffffa2e2a6eb, in 
    #26   Object "/lib/aarch64-linux-gnu/libruby-3.2.so.3.2", at 0xffffa2d52507, in rb_protect
    #25   Object "/lib/aarch64-linux-gnu/libruby-3.2.so.3.2", at 0xffffa2ef96f7, in rb_yield
    #24   Object "/lib/aarch64-linux-gnu/libruby-3.2.so.3.2", at 0xffffa2ef5237, in rb_vm_exec
    #23   Object "/lib/aarch64-linux-gnu/libruby-3.2.so.3.2", at 0xffffa2ef237f, in 
    #22   Object "/lib/aarch64-linux-gnu/libruby-3.2.so.3.2", at 0xffffa2eedd03, in 
    #21   Object "/lib/aarch64-linux-gnu/libruby-3.2.so.3.2", at 0xffffa2eeba87, in 
    #20   Object "/usr/lib/aarch64-linux-gnu/ruby/3.2.0/fiddle.so", at 0xffff9dc86467, in 
    #19   Object "/lib/aarch64-linux-gnu/libruby-3.2.so.3.2", at 0xffffa2eb3c77, in rb_nogvl
    #18   Object "/usr/lib/aarch64-linux-gnu/ruby/3.2.0/fiddle.so", at 0xffff9dc85ec3, in 
    #17   Object "/lib/aarch64-linux-gnu/libffi.so.8", at 0xffff9dc3383b, in 
    #16   Object "/lib/aarch64-linux-gnu/libffi.so.8", at 0xffff9dc36623, in 
    #15   Object "/opt/ros/jazzy/opt/gz_sim_vendor/lib/libgz-sim8-gz.so.8.3.0", at 0xffff9d00facb, in runServer
    #14   Object "/opt/ros/jazzy/opt/gz_sim_vendor/lib/libgz-sim8.so.8", at 0xffff9ccef17b, in gz::sim::v8::Server::Server(gz::sim::v8::ServerConfig const&)
    #13   Object "/opt/ros/jazzy/opt/gz_sim_vendor/lib/libgz-sim8.so.8", at 0xffff9ccfc787, in 
    #12   Object "/opt/ros/jazzy/opt/gz_sim_vendor/lib/libgz-sim8.so.8", at 0xffff9ccf9dff, in gz::sim::v8::SimulationRunner::SimulationRunner(sdf::v14::World const*, std::shared_ptr<gz::sim::v8::SystemLoader> const&, gz::sim::v8::ServerConfig const&)
    #11   Object "/opt/ros/jazzy/opt/gz_sim_vendor/lib/libgz-sim8.so.8", at 0xffff9ccb1c13, in gz::sim::v8::LevelManager::UpdateLevelsState()
    #10   Object "/opt/ros/jazzy/opt/gz_sim_vendor/lib/libgz-sim8.so.8", at 0xffff9ccaddf7, in gz::sim::v8::LevelManager::LoadActiveEntities(std::set<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)
    #9    Object "/opt/ros/jazzy/opt/gz_sim_vendor/lib/libgz-sim8.so.8", at 0xffff9ccd9817, in gz::sim::v8::SdfEntityCreator::CreateEntities(sdf::v14::Model const*)
    #8    Object "/opt/ros/jazzy/opt/gz_sim_vendor/lib/libgz-sim8.so.8", at 0xffff9ccad967, in 
    #7    Object "/opt/ros/jazzy/opt/gz_sim_vendor/lib/libgz-sim8.so.8", at 0xffff9cd01b9f, in gz::sim::v8::SimulationRunner::LoadPlugins(unsigned long, std::vector<sdf::v14::Plugin, std::allocator<sdf::v14::Plugin> > const&)
    #6    Object "/opt/ros/jazzy/opt/gz_sim_vendor/lib/libgz-sim8.so.8", at 0xffff9cd13abf, in gz::sim::v8::SystemManager::LoadPlugin(unsigned long, sdf::v14::Plugin const&)
    #5    Object "/opt/ros/jazzy/opt/gz_sim_vendor/lib/libgz-sim8.so.8", at 0xffff9cd135a3, in gz::sim::v8::SystemManager::AddSystemImpl(gz::sim::v8::SystemInternal, std::shared_ptr<sdf::v14::Element const>)
    #4    Object "/usr/lib/aarch64-linux-gnu/gz-sim-8/plugins/libgz-sim-python-system-loader-system.so", at 0xffff900675ef, in 
    #3    Object "/lib/aarch64-linux-gnu/libpython3.12.so.1.0", at 0xffff72873e43, in PyImport_ImportModule
    #2    Object "/lib/aarch64-linux-gnu/libpython3.12.so.1.0", at 0xffff727e39ef, in 
    #1    Object "/lib/aarch64-linux-gnu/libpython3.12.so.1.0", at 0xffff727c576f, in PyUnicode_New
    #0    Object "/lib/aarch64-linux-gnu/libpython3.12.so.1.0", at 0xffff7278de24, in 
    Segmentation fault (Address not mapped to object [0x10])
    ^C/opt/ros/jazzy/opt/gz_sim_vendor/lib/ruby/gz/cmdsim8.rb:201:in `kill': No such process (Errno::ESRCH)
	    from /opt/ros/jazzy/opt/gz_sim_vendor/lib/ruby/gz/cmdsim8.rb:201:in `killProcess'
	    from /opt/ros/jazzy/opt/gz_sim_vendor/lib/ruby/gz/cmdsim8.rb:569:in `block in execute'
	    from /opt/ros/jazzy/opt/gz_sim_vendor/lib/ruby/gz/cmdsim8.rb:217:in `sleep'
	    from /opt/ros/jazzy/opt/gz_sim_vendor/lib/ruby/gz/cmdsim8.rb:217:in `killProcess'
	    from /opt/ros/jazzy/opt/gz_sim_vendor/lib/ruby/gz/cmdsim8.rb:577:in `execute'
	    from /opt/ros/jazzy/opt/gz_tools_vendor/bin/gz:309:in `<main>'
   ```
FYI, [bhimanyubhowmik](https://github.com/IOES-Lab/dave/commits?author=abhimanyubhowmik)

I will continue to wrestle with this but Please help! [azeey](https://github.com/azeey)

azeey, 2024-08-06:
Sorry to see you ran into this. Unfortunately python bindings aren't installed with the vendor packages yet (https://github.com/gazebo-tooling/gz_vendor/issues/2). If you don't need to use this with ROS, you can just install `gz-harmonic` and run your code without sourcing the ROS workspace. If you do need to use ROS, please follow https://gazebosim.org/docs/latest/ros2_gz_vendor_pkgs/#installing-non-default-gazebo-ros-2-pairings-with-vendor-packages

azeey, 2024-08-06:
I'll go ahead and close this since it's tracked by the gz_vendor issue.

closed 2024-08-06 by azeey

woensug-choi, 2024-08-07:
@azeey Confirmed that running without sourcing ROS didn't cause any problem. I am trying to us ROS2 Jazzy with Gazebo Harmonic (which is the default pair). Binary ROS2 install and Source install of Gazebo Harmonic didn't cause such error. Installing both Jazzy and Harmonic with binary installation method did. Would it be solved later?

EceChaik, 2024-09-27:
I'm facing the exact same issue while trying to replicate the CartPole demo but using ROS2 Jazzy.
I'm struggling to understand what the problem is that leads to the segmentation fault being thrown, since I have all the ros-jazzy-gz-* packages installed. Is it not possible to use a custom plugin with Gazebo Harmonic and ROS2 Jazzy (which is the default pairing btw)?