gazebosim/gz-sim #2520: Spring-damper prismatic joint system plugin

damanikjosh, 2024-08-07:
## Desired behavior

Simulating spring-damper physics on prismatic joint. [https://en.wikipedia.org/wiki/Mass-spring-damper_model](https://en.wikipedia.org/wiki/Mass-spring-damper_model). In Gazebo classic it was implemented by modifying ODE parameters [https://github.com/arpg/Gazebo/blob/master/test/worlds/spring_damper_test.world](https://github.com/arpg/Gazebo/blob/master/test/worlds/spring_damper_test.world), but I can't get it to work on gz-sim. A similar issue was posted in #1961, but instead of modifying the physics, I suggest adding a system plugin to simulate this.

## Implementation suggestion

1. Create `JointPosition` and `JointVelocity` for prismatic joint
2. Send `JointForceCmd` to the joint entity with spring-damper equation
3. Add plugin on sdf file similar like below
```xml
<plugin filename="libSpringSystem.so" name="spring_system::SpringSystem">
      <spring_constant>1200.0</spring_constant>
      <damping_constant>150.0</damping_constant>
      <joint_name>left_front_shock</joint_name>
      <joint_name>right_front_shock</joint_name>
      <joint_name>left_rear_shock</joint_name>
      <joint_name>right_rear_shock</joint_name>
</plugin>
```

azeey, 2024-08-07:
Have you tried using the `damping` and `spring_stiffness` parameters in the `<dynamics>` tag of a joint axis (http://sdformat.org/spec?ver=1.11&elem=joint#axis_dynamics)? I believe those work for prismatic joints, so I don't see the value of adding a separate system to simulate this.

damanikjosh, 2024-08-07:
I finally solved this. The problem was the model loaded using URDF and I found out that URDF doesn't support `spring_stiffness` parameter natively ([http://wiki.ros.org/urdf/XML/joint](http://wiki.ros.org/urdf/XML/joint)). But this one works ([https://github.com/gazebosim/sdf_tutorials/blob/master/urdf/sdf_extensions.md](https://github.com/gazebosim/sdf_tutorials/blob/master/urdf/sdf_extensions.md))
```xml
<gazebo>
    <springStiffness>1200</<springStiffness>
</gazebo>
```

I will close this issue, thank you.

closed 2024-08-07 by damanikjosh