gazebosim/gz-sim #2522: Gazebo goes not responding after loading the window

aadith-warrier, 2024-08-12:
## Environment
* OS Version: Ubuntu 24.04 LTS x86_64
* Source or binary build? Binary v8.5.0
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre | ogre2]. tried both orge and ogre2
      - [ ] Sensor rendering error. 
      - [ ] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [x] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [x] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [x] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          echo "$DISPLAY"
          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          ps aux | grep Xorg
          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```
      - On Windows, run `dxdiag` and report the GPU-related information.
      - On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. Copy the output here.
        
    - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering` 

<details>

```
LANG=C lspci -nn | grep VGA
00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 530 [8086:1912] (rev 06)
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GM107 [GeForce GTX 745] [10de:1382] (rev a2)
```

```
echo "$DISPLAY"
:1
```

```
LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'
direct rendering: Yes
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: NVIDIA GeForce GTX 745/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 535.183.01
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6.0 NVIDIA 535.183.01
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 535.183.01
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```

```
ps aux | grep Xorg
abc-i9     17972  4.2  0.4 8299636 158812 tty2   Sl+  15:16   1:21 /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1000/gdm/Xauthority -nolisten tcp -background none -noreset -keeptty -novtswitch -verbose 3
abc-i9     21753  0.0  0.0   9468  2176 pts/2    S+   15:48   0:00 grep --color=auto Xorg
```

```
sudo env LANG=C X -version
[sudo] password for abc-i9: 

X.Org X Server 1.21.1.11
X Protocol Version 11, Revision 0
Current Operating System: Linux abci9-OptiPlex-7040 6.8.0-40-generic #40-Ubuntu SMP PREEMPT_DYNAMIC Fri Jul  5 10:34:03 UTC 2024 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.8.0-40-generic root=UUID=1965d466-bc73-4cf0-88eb-604aea38dff2 ro quiet splash vt.handoff=7
xorg-server 2:21.1.12-1ubuntu1 (For technical support please see http://www.ubuntu.com/support) 
Current version of pixman: 0.42.2
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
```

```
~/.gz/rendering does not exist
```

</details>

## Description
* Expected behavior: Gazebo launches with an empty world
* Actual behavior: Gazebo windows opens and freezes then goes not responding

## Steps to reproduce

1. Run ```gz sim```

## Output

```
OptiPlex-7040:~$ gz sim
Escalating to SIGKILL on [Gazebo Sim Server]
abc-i9@abci9-OptiPlex-7040:~$ gz sim --verbose
[Msg] Gazebo Sim GUI    v8.5.0
[Msg] Gazebo Sim Quick start dialog
[GUI] [Msg] Not showing Quick start menu.
[GUI] [Wrn] [Gui.cc:271] Waited for 10s for a subscriber to [/gazebo/starting_world] and got none.
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:413:17: QML ToolButton: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:413:17: QML ToolButton: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:413:17: QML ToolButton: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:908] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:309:21: QML Button: Binding loop detected for property "implicitHeight"
Escalating to SIGKILL on [Gazebo Sim Server]
```
```gz sim --render-engine ogre``` fails similarly

[image]

iche033, 2024-08-12:
can you try with `-v 4` instead of `--verbose`, e.g. 

```
gz sim shapes.sdf --verbose
```

and see if it prints some error output

azeey, 2024-08-12:
Also, this might be related to a network configuration issue. See https://github.com/gazebosim/gz-transport/issues/476#issuecomment-2282728388

aadith-warrier, 2024-08-13:
This was a network configuration issue as mentioned in https://github.com/gazebosim/gz-transport/issues/476#issuecomment-2282728388.

Adding this to https://gazebosim.org/docs/latest/troubleshooting/ would help.

Thank you.

closed 2024-08-13 by aadith-warrier

azeey, 2024-08-13:
@aadith-warrier, I'm glad that was the issue. I agree it would be good to add it to the troubleshooting page. Would you be able to create a PR to https://github.com/gazebosim/docs/edit/master/harmonic/troubleshooting.md?

aadith-warrier, 2024-08-14:
Yeah sure. https://github.com/gazebosim/docs/pull/484