gazebosim/gz-sim #2532: Setting multiple light properties using  `sim::Light` doesn't work properly

azeey, 2024-08-21:
## Environment
* OS Version: all
* Source or binary build? 8.6.0

## Description
* Expected behavior: Calling `sim::Light::SetDiffuseColor` followed by `sim::Light::SetSpecularColor` to work. Example:
```cpp
gz::sim::Light light(light_entity);
light.SetDiffuseColor(ecm, gz::math::Color(1.0, 0.0, 0.0, 0.0));
light.SetSpecularColor(ecm, gz::math::Color(1.0, 0.0, 0.0, 0.0));
```
* Actual behavior: Only the last set value takes effect. Looking at the code in https://github.com/gazebosim/gz-sim/blob/523b01b390ed9572522c99f792df01f583bb579f/src/Light.cc#L326-L344 calling `SetDiffuseColor` or `SetSpecularColor`  will set the `LightCmd` component.  But if you call one after the other without a whole step in between, the `LightCmd` from one will overwrite the other. The code already checks if there's a `LightCmd` component, but instead of updating the value of the component, it always replaces it with a new value.

shr-eyas, 2024-10-28:
hi, i would like to contribute.

azeey, 2024-10-28:
Thanks @shr-eyas ! I'll go ahead and assign this to you.

shr-eyas, 2024-10-29:
@azeey thank you! i have been an avid user of gazebo and ros for past 2 years and this would be my first open-source contribution. 
i have read the guidelines but it would be great if you can share some resources to help me get started.

shr-eyas, 2024-10-29:
hi @azeey , i believe i have solved the issue but as this is my first contribution, i am facing some difficulties with how to fulfill the prerequisites for creating a PR. specifically, with the tests. can you please help/guide me with the same?

shr-eyas, 2024-10-29:
hi, @azeey ! i believe i have solved the issue however, i am having issues testing it. this is my first open-source contribution and it would be great if you can provide some help/guidance.

Shreshtha2000, 2026-08-30:
Hi @azeey and @shr-eyas — I noticed this has PR #2660 linked but it looks like it stalled around Jan 2025. Just wanted to check if anyone is still actively working on this, or would it be okay for me to pick it up? Happy to build on the existing PR if that work is still usable. This would be my first contribution, so any pointers are welcome too.

azeey, 2026-08-31:
Hi @Shreshtha2000, thanks for your interest in this. Could you do a review of #2660? If you do some local testing and report back, that would be super helpful.

Shreshtha2000, 2026-09-01:
Sure, give me some time, will test it out.

Shreshtha2000, 2026-09-06:
Hi @azeey I tested out #2660 , and it does seem to fix some part of the issue by manipulating the else block (which seems correct), as when both setters called one after the other, the color properties did change, but there is another issue that I found, which is as follows:
if (!lightCmd) branch (which runs whenever this is the first commanded change to a light) in both setSpecularColor and setDiffuseColor in Light.cc, still seeds from a blank, default constructed msgs::Light. That means whichever setter happens to create the LightCmd first still zeroes out intensity, direction, attenuation, etc. Thus when i was testing out the setter commands the intensity of light was dropping to zero.
Do let me know if it is a genuine issue or maybe there was some steps that I took wrong in my testing.
I am also working on the fix for the issue, let me know if you are interested in that.
I was using ubuntu 22.04 and gz-sim8 branch for testing.

Shreshtha2000, 2026-09-12:
Hi @azeey, for the issue that I faced, I am proposing a fix for it , if the lightCMD is not found, we fetch the current light data and create a light message using that only, do check out my commit, also I have included the change from #2660 for now
https://github.com/Shreshtha2000/gz-sim/commit/da3aed422effd1117fb7570b9cdf58c19ee553da
Also let me know if I need to look more into it.