gazebosim/gz-sim #2543: Crash after calling /optical_tactile_sensor/enable over OpticalTactilePlugin

zhileiren, 2024-08-25:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build?
    source build
    gz-sim version: d662000273ae88e041ee30ec8d24b5e3beed6db7
    built with 
    gcc version: 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04) 
    build options: -DCMAKE_BUILD_TYPE=Coverage

## Description
* Expected behavior: Gazebo doesn't crash
* Actual behavior: Gazebo crashes

## Steps to reproduce
1. gz sim [a.txt](https://github.com/user-attachments/files/16739055/a.txt) -r # which is actually gz-sim/test/worlds/optical_tactile_plugin.sdf
2. call /optical_tactile_sensor/enable service, with

```bash
gz service --timeout 10000 -s /optical_tactile_sensor/enable --reptype gz.msgs.Empty --reqtype gz.msgs.Boolean --req ''
```
## Output
```
Stack trace (most recent call last) in thread 141106:
#23   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#22   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ff845f8084f, in 
#21   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ff845eeeac2, in 
#20   Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7ff841c2b252, in 
#19   Object "/home/ren/play/robot/workspace/install/lib/libgz-transport13.so.13", at 0x7ff83c5f3e3d, in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (gz::transport::v13::NodeShared::*)(), gz::transport::v13::NodeShared*> > >::_M_run()
#18   Object "/home/ren/play/robot/workspace/install/lib/libgz-transport13.so.13", at 0x7ff83c5f3f4d, in std::thread::_Invoker<std::tuple<void (gz::transport::v13::NodeShared::*)(), gz::transport::v13::NodeShared*> >::operator()()
#17   Object "/home/ren/play/robot/workspace/install/lib/libgz-transport13.so.13", at 0x7ff83c5f4232, in void std::thread::_Invoker<std::tuple<void (gz::transport::v13::NodeShared::*)(), gz::transport::v13::NodeShared*> >::_M_invoke<0ul, 1ul>(std::_Index_tuple<0ul, 1ul>)
#16   Object "/home/ren/play/robot/workspace/install/lib/libgz-transport13.so.13", at 0x7ff83c5f47d2, in std::__invoke_result<void (gz::transport::v13::NodeShared::*)(), gz::transport::v13::NodeShared*>::type std::__invoke<void (gz::transport::v13::NodeShared::*)(), gz::transport::v13::NodeShared*>(void (gz::transport::v13::NodeShared::*&&)(), gz::transport::v13::NodeShared*&&)
#15   Object "/home/ren/play/robot/workspace/install/lib/libgz-transport13.so.13", at 0x7ff83c5f4cc2, in void std::__invoke_impl<void, void (gz::transport::v13::NodeShared::*)(), gz::transport::v13::NodeShared*>(std::__invoke_memfun_deref, void (gz::transport::v13::NodeShared::*&&)(), gz::transport::v13::NodeShared*&&)
#14   Object "/home/ren/play/robot/workspace/install/lib/libgz-transport13.so.13", at 0x7ff83c589e5c, in gz::transport::v13::NodeShared::RunReceptionTask()
#13   Object "/home/ren/play/robot/workspace/install/lib/libgz-transport13.so.13", at 0x7ff83c58e709, in gz::transport::v13::NodeShared::RecvSrvRequest()
#12   Object "/data/play/robot/workspace/install/lib/gz-sim-8/plugins/libgz-sim-opticaltactileplugin-system.so", at 0x7ff8014905e4, in gz::transport::v13::RepHandler<gz::msgs::Boolean, gz::msgs::Empty>::RunCallback(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)
#11   Object "/data/play/robot/workspace/install/lib/gz-sim-8/plugins/libgz-sim-opticaltactileplugin-system.so", at 0x7ff8014949b2, in std::function<bool (gz::msgs::Boolean const&, gz::msgs::Empty&)>::operator()(gz::msgs::Boolean const&, gz::msgs::Empty&) const
#10   Object "/data/play/robot/workspace/install/lib/gz-sim-8/plugins/libgz-sim-opticaltactileplugin-system.so", at 0x7ff80143273f, in std::_Function_handler<bool (gz::msgs::Boolean const&, gz::msgs::Empty&), gz::transport::v13::Node::Advertise<gz::sim::v8::systems::OpticalTactilePluginPrivate, gz::msgs::Boolean>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, void (gz::sim::v8::systems::OpticalTactilePluginPrivate::*)(gz::msgs::Boolean const&), gz::sim::v8::systems::OpticalTactilePluginPrivate*, gz::transport::v13::AdvertiseServiceOptions const&)::{lambda(gz::msgs::Boolean const&, gz::msgs::Empty&)#1}>::_M_invoke(std::_Any_data const&, gz::msgs::Boolean const&, gz::msgs::Empty&)
#9    Object "/data/play/robot/workspace/install/lib/gz-sim-8/plugins/libgz-sim-opticaltactileplugin-system.so", at 0x7ff80144299d, in std::enable_if<is_invocable_r_v<bool, gz::transport::v13::Node::Advertise<gz::sim::v8::systems::OpticalTactilePluginPrivate, gz::msgs::Boolean>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, void (gz::sim::v8::systems::OpticalTactilePluginPrivate::*)(gz::msgs::Boolean const&), gz::sim::v8::systems::OpticalTactilePluginPrivate*, gz::transport::v13::AdvertiseServiceOptions const&)::{lambda(gz::msgs::Boolean const&, gz::msgs::Empty&)#1}&, gz::msgs::Boolean const&, gz::msgs::Empty&>, bool>::type std::__invoke_r<bool, gz::transport::v13::Node::Advertise<gz::sim::v8::systems::OpticalTactilePluginPrivate, gz::msgs::Boolean>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, void (gz::sim::v8::systems::OpticalTactilePluginPrivate::*)(gz::msgs::Boolean const&), gz::sim::v8::systems::OpticalTactilePluginPrivate*, gz::transport::v13::AdvertiseServiceOptions const&)::{lambda(gz::msgs::Boolean const&, gz::msgs::Empty&)#1}&, gz::msgs::Boolean const&, gz::msgs::Empty&>(gz::transport::v13::Node::Advertise<gz::sim::v8::systems::OpticalTactilePluginPrivate, gz::msgs::Boolean>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, void (gz::sim::v8::systems::OpticalTactilePluginPrivate::*)(gz::msgs::Boolean const&), gz::sim::v8::systems::OpticalTactilePluginPrivate*, gz::transport::v13::AdvertiseServiceOptions const&)::{lambda(gz::msgs::Boolean const&, gz::msgs::Empty&)#1}&, gz::msgs::Boolean const&, gz::msgs::Empty&)
#8    Object "/data/play/robot/workspace/install/lib/gz-sim-8/plugins/libgz-sim-opticaltactileplugin-system.so", at 0x7ff80144ed81, in bool std::__invoke_impl<bool, gz::transport::v13::Node::Advertise<gz::sim::v8::systems::OpticalTactilePluginPrivate, gz::msgs::Boolean>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, void (gz::sim::v8::systems::OpticalTactilePluginPrivate::*)(gz::msgs::Boolean const&), gz::sim::v8::systems::OpticalTactilePluginPrivate*, gz::transport::v13::AdvertiseServiceOptions const&)::{lambda(gz::msgs::Boolean const&, gz::msgs::Empty&)#1}&, gz::msgs::Boolean const&, gz::msgs::Empty&>(std::__invoke_other, gz::transport::v13::Node::Advertise<gz::sim::v8::systems::OpticalTactilePluginPrivate, gz::msgs::Boolean>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, void (gz::sim::v8::systems::OpticalTactilePluginPrivate::*)(gz::msgs::Boolean const&), gz::sim::v8::systems::OpticalTactilePluginPrivate*, gz::transport::v13::AdvertiseServiceOptions const&)::{lambda(gz::msgs::Boolean const&, gz::msgs::Empty&)#1}&, gz::msgs::Boolean const&, gz::msgs::Empty&)
#7    Object "/data/play/robot/workspace/install/lib/gz-sim-8/plugins/libgz-sim-opticaltactileplugin-system.so", at 0x7ff801418a2e, in gz::transport::v13::Node::Advertise<gz::sim::v8::systems::OpticalTactilePluginPrivate, gz::msgs::Boolean>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, void (gz::sim::v8::systems::OpticalTactilePluginPrivate::*)(gz::msgs::Boolean const&), gz::sim::v8::systems::OpticalTactilePluginPrivate*, gz::transport::v13::AdvertiseServiceOptions const&)::{lambda(gz::msgs::Boolean const&, gz::msgs::Empty&)#1}::operator()(gz::msgs::Boolean const&, gz::msgs::Empty&) const
#6    Object "/data/play/robot/workspace/install/lib/gz-sim-8/plugins/libgz-sim-opticaltactileplugin-system.so", at 0x7ff801421cf7, in void std::_Bind<void (gz::sim::v8::systems::OpticalTactilePluginPrivate::*(gz::sim::v8::systems::OpticalTactilePluginPrivate*, std::_Placeholder<1>))(gz::msgs::Boolean const&)>::operator()<gz::msgs::Boolean const&, void>(gz::msgs::Boolean const&)
#5    Object "/data/play/robot/workspace/install/lib/gz-sim-8/plugins/libgz-sim-opticaltactileplugin-system.so", at 0x7ff8014325a3, in void std::_Bind<void (gz::sim::v8::systems::OpticalTactilePluginPrivate::*(gz::sim::v8::systems::OpticalTactilePluginPrivate*, std::_Placeholder<1>))(gz::msgs::Boolean const&)>::__call<void, gz::msgs::Boolean const&, 0ul, 1ul>(std::tuple<gz::msgs::Boolean const&>&&, std::_Index_tuple<0ul, 1ul>)
[... cut ...]
#1    Object "/data/play/robot/workspace/install/lib/gz-sim-8/plugins/libgz-sim-opticaltactileplugin-system.so", at 0x7ff8014adc71, in gz::sim::v8::systems::optical_tactile_sensor::OpticalTactilePluginVisualization::RemoveNormalForcesAndContactsMarkers()
#0    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7ff841c9dd85, in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > std::operator+<char, std::char_traits<char>, std::allocator<char> >(char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
Segmentation fault (Address not mapped to object [0x18])
```

zhileiren, 2024-08-27:
I just noticed that only `/optical_tactile_sensor/enable` is listed in `gz service -l`. Instead, for another sdf file gz-sim/examples/worlds/optical_tactile_sensor_plugin.sdf (mentioned in #2541), both `/optical_tactile_sensor/enable`  and `/optical_tactile_plugin/enable` are published. Over this file, no crash appears. I'm not sure if it's the case?

zhileiren, 2024-08-27:
BTW, for gz-sim/examples/worlds/optical_tactile_sensor_plugin.sdf, the model is not rendered in my environment, but not sure it's relevant.

[image]

iche033, 2024-11-11:
I must be running the wrong world file before. I can now reproduce it and created a fix: https://github.com/gazebosim/gz-sim/pull/2674

closed 2024-11-12 by ahcorde