gazebosim/gz-sim #2555: Console errors from gz-sim9 ubuntu noble CI tests mark build as unstable

iche033, 2024-08-29:
## Environment
* OS Version: 24.04
* Source or binary build? gz-sim9 on jenkins CI
 
## Description
* Expected behavior: test console log should not be counted as warnings on ubuntu noble CI
* Actual behavior: CI should be green and console log error / warnings should not mark CI as unstable

For example, [gz_sim-ci-gz-sim9-noble-amd64](https://build.osrfoundation.org/job/gz_sim-ci-gz-sim9-noble-amd64/1/) reports 49 new warnings. Closer inspection reveals that these are console output from tests. 

Here's an example test console output errors that are marked as warnings:

```
3: Running main() from /home/jenkins/workspace/gz_sim-ci-gz-sim9-noble-amd64/gz-sim/test/gtest_vendor/src/gtest_main.cc
53: [==========] Running 1 test from 1 test suite.
53: [----------] Global test environment set-up.
53: [----------] 1 test from MsgManagerTest
53: [ RUN      ] MsgManagerTest.MsgManager
53: (2024-08-29 14:40:56.902) [error] AddSubscriber() error: Address already attached to a different model
53: (2024-08-29 14:40:56.902) [error] RemoveInbound() error: Unable to find address [not_found]
53: (2024-08-29 14:40:56.902) [error] RemoveOutbound() error: Unable to find address [not_found]
53: [       OK ] MsgManagerTest.MsgManager (1 ms)
53: [----------] 1 test from MsgManagerTest (1 ms total)
53: 
53: [----------] Global test environment tear-down
53: [==========] 1 test from 1 test suite ran. (1 ms total)
53: [  PASSED  ] 1 test.
```

j-rivero, 2024-08-30:
Indeed, looks like a false positive detected by the GCC Warning plugin. Need to find a workaround to exclude these lines.

azeey, 2024-08-30:
> Need to find a workaround to exclude these lines.

Can we limit the plugin that parses the output to only look at the logs up to and including the compile step, but not the test step?

j-rivero, 2024-08-30:
> > 
> 
> Can we limit the plugin that parses the output to only look at the logs up to and including the compile step, but not the test step?

I'm afraid that there is no easy way for the Jenkins plugins to determine parts of a build script semantically. Not very nice but I'm trying to exclude warnings by file pattern since the errors use the form of: `digit+: (` which is a file name that we probably are not going to find easily.

j-rivero, 2024-08-30:
> Not very nice but I'm trying to exclude warnings by file pattern since the errors use the form of: digit+: ( which is a file name that we probably are not going to find easily.

Tested here: https://build.osrfoundation.org/job/_test_gz_sim-ci-gz-sim9-noble-amd64/

closed 2024-08-30 by azeey